adafruit/Adafruit-GFX-Library #46: drawBitmap PROGMEM constraint

TheNotary, 2015-05-07:
I'm trying to define a class that has a draw method which draws a bitmap to the matrix.  But when I run that function, I get very odd, though consistent results.  Upon reviewing the code, it's indicated 

>> [drawBitmap] ... Draw a 1-bit color bitmap at the specified x, y position from the provided bitmap buffer (must be PROGMEM memory)...

I'm still new to arduino programming, so I have to ask why does it only work with data in PROGMEM?

TheNotary, 2015-05-08:
I'm uncovering oddities related to the backpack library not this one I think...  I'm trying to call drawBitmap() and displayWrite() from within class methods.  But it seams displayWrite() may have a built-in displayClear() command that's run.  

Feel free to close this, it may be in the wrong place.  I'll keep working at it till I can figure out the best place to raise this issue and either close this issue or re-articulate myself when I have a better understanding of the issue's nature.

TheNotary, 2015-05-11:
Hey, I wrote a function that draws a bitmap to the screen from 'regular memory'.  This allows for dynamically generating bitmaps to be rendered and opens the doors to more possibilities for using these 8x8 matrices.  

If I cleaned up my code would you like a pull request?  I'm kinda new so I might be missing something about working with PROGMEM versus dynamically generated content.  

Here's my work, but note that I don't entirely grasp this line `pgm_read_byte(bitmap + j * byteWidth + i / 8) & (128 >> (i & 7)))` and so my code lacks parallelism with the existing PROGMEM code and doesn't use the byteWidth variable.   

    void Adafruit_GFX::drawBitmap(int16_t x, int16_t y,
			          uint8_t *bitmap, int16_t w, int16_t h,
			          uint16_t color) {
      
      int16_t i, j, byteWidth = (w + 7) / 8;

      for(j=0; j<h; j++) {
        for(i=0; i<w; i++ ) {
          if(bitRead(bitmap[j], 7-i))  
            drawPixel(x+i, y+j, color);
        }
      }
    }

    // Draw a 1-bit color bitmap at the specified x, y position from the
    // provided bitmap buffer (must not be PROGMEM memory nor a const) 
    // using color as the foreground color and bg as the background color.
    void Adafruit_GFX::drawBitmap(int16_t x, int16_t y,
			          uint8_t *bitmap, int16_t w, int16_t h,
			          uint16_t color, uint16_t bg) {

      int16_t i, j, byteWidth = (w + 7) / 8;

      for(j=0; j<h; j++) {
        for(i=0; i<w; i++ ) {
          if(bitRead(bitmap[j], 7-i))  
            drawPixel(x+i, y+j, color);
          else
            drawPixel(x+i, y+j, bg);
        }
      }
    }

Also, notice how those two funcs are two seperate funcs.  They could be merged to 1 func that simply calls the other func if we had the one that doesn't specify any color to call the func that does specify the color, and pass in LED_OFF.  The only loss is if ppl are using the function to drawBitmap without erasing the clearing the grid beforehand which I feel may be unlikely.

TheNotary, 2015-05-11:
Dropped a pull request #47 the diff tells the story better than my rambling here =/

fingolfin, 2016-01-04:
Since #47 was merged, this issue should be resolved, shouldn't it?

closed 2016-01-04 by TheNotary