adafruit/Adafruit-GFX-Library #48: ESP8266 compatibility

somhi, 2015-05-26:
For the Adafruit_GFX library to work with the ESP8266 board I have inserted the line "#elif defined ESP8266" in the Adafruit_GFX.cpp file. Otherwise also works commenting the line //#define pgm_read_byte(addr)...

Code in Adafruit_GFX.cpp should look like this:
#ifdef __AVR__
 #include <avr/pgmspace.h>
#elif defined ESP8266    //Added for compatibility with ESP8266 BOARD
#else
 #define pgm_read_byte(addr) (*(const unsigned char *)(addr))
#endif

I've created the ESP_SSD1306 library which is a copy of the Adafruit_SSD1306 library with some modifications in order to work with the ESP8266 board.

See more details at   https://github.com/somhi/ESP_SSD1306

MStylesMS, 2016-08-02:
I'm trying to connect one of the Adafruit 7-segment LED's to an ESP8266 v3 but can't get it to work.  I have SCL connected to GPIO14 and and SDA connected to GPIO12, along with GND to ground and VCC to +3v.  Any idea on where I'm missing it?

GerardHaagh, 2017-10-21:
Change still needed! pgm_read_byte is redefined by the macro(s) at line Adafruit_GFX.cpp:47, overruling the pgm_read_byte from the platform library for the ESP8266. This causes exceptions to be thrown (ie when printing text to the TFT).

Bzzz, 2018-03-17:
Got any "working" example code of that? Started running a few examples (built-in and from the interwebs) with a WEMOS Mini (ESP8266) and some e-ink display - no warnings, no errors.

mikeysklar, 2026-10-21:
The library now builds for ESP8266 in CI on every change.
It uses the platform's own pgm_read_byte, so the old redefinition no longer applies.
I tested built-in and FreeSans fonts on a Feather HUZZAH ESP8266 today.
Closing this one.
Please reopen if you still see it on the current release.

closed 2026-10-21 by mikeysklar