adafruit/Adafruit-GFX-Library #52: How do people most commonly use/bundle Adafruit-GFX in projects?

yyyc514, 2015-06-19:
Do they install it as a library from the Arduino IDE?  And then when you distribute your source do you list in the README everything they need to install the library or do you vendor it locally with your own code?

I'm struggling with this since the libraries I write are often tons of little files to separate everything nicely... and that's much harder to vendor than one big library in two files (since the IDE provides no good way to put things in subdirectories that I've found).

Plus if Arduboy was to decide to use Adafruit-GFX we'd then had to deal with this exact problem.  Tell people to add the library vs vendor in our own source tree...

Thoughts?

TheNotary, 2015-08-04:
You must be coming from a language with a sofisticated package manager.  I use platformio for building (not an IDE).  My code structure looks like this

    ├── changelog
    ├── lib
    │   ├── Adafruit-RGB-LCD-Shield-Library-master
    │   ├── TinyAdafruit_RGBLCDShield
    │   ├── TinyRTCLib-master
    │   └── TinyWireM-master
    ├── platformio.ini
    ├── README.md
    └── src
        └── lcd_trinket_test.ino

So my actual code is in src, but my externally sourced libraries are in the lib folder.  How do I get them there?  I just git clone them in, and then delete the .git file so they can be tracked into my repo normally.  When I build with platformio, it just automatically checks that folder and builds with the stuff I pull into lib/  

Btw, adafruit forums is often a pretty cool place.  forums.adafruit.com/

mikeysklar, 2026-10-21:
The README says the Arduino Library Manager can install GFX.

That also pulls in its dependencies, like Adafruit BusIO.

For general how-to questions, https://forums.adafruit.com is the best place.

Closing this one.

closed 2026-10-21 by mikeysklar