adafruit/Adafruit-GFX-Library #53: Compiler warnings on ESP8266

tdicola, 2015-06-21:
It looks like there are some warnings about PROGMEM and flash memory functions being redefined:
```
C:\Users\Mike\AppData\Local\Temp\build8213996557431336517.tmp\Adafruit_GFX\Adafruit_GFX.cpp.o 
In file included from F:\Code\ArduinoCode\libraries\Adafruit_GFX\Adafruit_GFX.cpp:35:0:
F:\Code\ArduinoCode\libraries\Adafruit_GFX\glcdfont.c:8:0: warning: "PROGMEM" redefined [enabled by default]
  #define PROGMEM
 ^
In file included from C:\Users\Mike\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.4-673-g8cd3697\cores\esp8266/Arduino.h:38:0,
                 from F:\Code\ArduinoCode\libraries\Adafruit_GFX\Adafruit_GFX.h:5,
                 from F:\Code\ArduinoCode\libraries\Adafruit_GFX\Adafruit_GFX.cpp:34:
C:\Users\Mike\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.4-673-g8cd3697\cores\esp8266/pgmspace.h:15:0: note: this is the location of the previous definition
 #define PROGMEM     ICACHE_RODATA_ATTR
 ^
F:\Code\ArduinoCode\libraries\Adafruit_GFX\Adafruit_GFX.cpp:39:0: warning: "pgm_read_byte" redefined [enabled by default]
  #define pgm_read_byte(addr) (*(const unsigned char *)(addr))
 ^
In file included from C:\Users\Mike\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.4-673-g8cd3697\cores\esp8266/Arduino.h:38:0,
                 from F:\Code\ArduinoCode\libraries\Adafruit_GFX\Adafruit_GFX.h:5,
                 from F:\Code\ArduinoCode\libraries\Adafruit_GFX\Adafruit_GFX.cpp:34:
C:\Users\Mike\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.4-673-g8cd3697\cores\esp8266/pgmspace.h:63:0: note: this is the location of the previous definition
 #define pgm_read_byte(addr)                               
```
Looks like we need to slightly tweak the AVR architecture checks to also handle the ESP8266.  Opening this as an issue so we don't lose track of it.

Makuna, 2017-04-03:
Is this still an issue?  I reworked the PROGMEM support for esp8266 in July 2015 to be more Arduino compatible and I haven't seem much issues with that since then.

PaintYourDragon, 2017-04-03:
Zero warnings here. Calling this closed.

closed 2017-04-03 by PaintYourDragon