adafruit/Adafruit-GFX-Library #55: Can I use another pin because need to use 4 device on 1 arduino

tonpp, 2015-07-30:
first thank you for library. and I have question about this.How can I use digit pin  because I need to use 4 OLED on 1 arduino
THank you.

daniel3514, 2015-07-31:
Saludos, gracias por tu interes
Lamentablemente no se puede usar con solo 4 pin, pero se puede configurar para usar con SPI 
https://learn.adafruit.com/adafruit-3-5-color-320x480-tft-touchscreen-breakout/spi-wiring-and-test

daniel3514, 2015-07-31:
Greetings, thank you for your interest
Unfortunately you can not wear with only 4 pin, but can be configured for use with SPI
https://learn.adafruit.com/adafruit-3-5-color-320x480-tft-touchscreen-breakout/spi-wiring-and-test

mikeysklar, 2026-10-21:
Pin setup is handled by the display driver, not GFX.
Running several displays on one board depends on the driver and its wiring.
Please ask on https://forums.adafruit.com with your display model and wiring.
Closing this one.

closed 2026-10-21 by mikeysklar