adafruit/Adafruit-GFX-Library #58: can you add raspberry pi support?

FishTest, 2015-09-08:
I want to use the library with ssd1306 library on raspberry pi ,but unfortunalty there's no raspberry pi version,i have to port it to raspberry pi with wiringpi ,but there's many many errors while compiling...
can you add raspberry pi support?

tdicola, 2015-09-22:
Check out this tutorial, I think it's just what you're looking for: https://learn.adafruit.com/ssd1306-oled-displays-with-raspberry-pi-and-beaglebone-black/overview  It shows how to use python and the python imaging library (which is similar in functionality to the GFX library) on the Pi with a SSD1306.

closed 2015-09-22 by tdicola

nkolban, 2015-12-16:
I understand about the Python library, but I was hoping for a C library?