adafruit/Adafruit-GFX-Library #66: Using fonts on the ESP8266

sticilface, 2016-01-01:
I'm running adafruit gfx just fine with the standard font.  However when i include fonts i get a stack trace. 

```
Exception (28):
epc1=0x4021131d epc2=0x00000000 epc3=0x00000000 excvaddr=0x000046fc depc=0x00000000

ctx: cont 
sp: 3fff3470 end: 3fff3750 offset: 01a0

>>>stack>>>
3fff3610:  4000050c 00000030 00000010 ffffffff  
3fff3620:  3fff61e0 00000000 3ffead13 40202764  
3fff3630:  3fff3660 00000000 3fff4580 4023b45c  
3fff3640:  3d96cd03 00000000 3b020821 ffffff82  
3fff3650:  3fff14f4 3fff4580 3fff56e0 40202141  
3fff3660:  3fff4580 00000000 00000000 40216ae1  
3fff3670:  3fff56e0 3fff4580 3fff56e0 4023b4e8  
3fff3680:  00000013 3fdf9f9f 3fff56e0 402189dc  
3fff3690:  3fff5ba0 3f800000 40216aa0 402172d0  
3fff36a0:  00000012 00000000 00000000 00000012  
3fff36b0:  000003fc 00000001 3ffead12 3fff2724  
3fff36c0:  3fff4580 000000aa 3fff14f4 40218ad1  
3fff36d0:  3f000012 00000013 3ffffc00 40202141  
3fff36e0:  3fffdad0 00000000 3fff62c0 4021a2c8  
3fff36f0:  4021d3c7 00000000 3fff1700 40218b36  
3fff3700:  3fff3e68 00000000 3fff62c0 40218b7b  
3fff3710:  00000000 00000000 3fff14f4 402101a8  
3fff3720:  3fffdad0 00000000 3fff271d 40217ced  
3fff3730:  3fffdad0 00000000 3fff271d 40202954  
3fff3740:  feefeffe feefeffe 3fff2730 402049bc  
<<<stack<<<
```

The causative function

```
40211178 <_ZN12Adafruit_GFX5writeEh>:
40211178:	e0c112        	addi	a1, a1, -32
4021117b:	61c9      	s32i.n	a12, a1, 24
4021117d:	02cd      	mov.n	a12, a2
4021117f:	7228      	l32i.n	a2, a2, 28
40211181:	7109      	s32i.n	a0, a1, 28
40211183:	51d9      	s32i.n	a13, a1, 20
40211185:	745030        	extui	a5, a3, 0, 8
40211188:	062256        	bnez	a2, 402111ee <_ZN12Adafruit_GFX5writeEh+0x76>
4021118b:	0e9566        	bnei	a5, 10, 4021119d <_ZN12Adafruit_GFX5writeEh+0x25>
4021118e:	091c32        	l16ui	a3, a12, 18
40211191:	180c42        	l8ui	a4, a12, 24
40211194:	b03430        	addx8	a3, a4, a3
40211197:	095c32        	s16i	a3, a12, 18
4021119a:	001286        	j	402111e8 <_ZN12Adafruit_GFX5writeEh+0x70>
4021119d:	d30c      	movi.n	a3, 13
4021119f:	029537        	bne	a5, a3, 402111a5 <_ZN12Adafruit_GFX5writeEh+0x2d>
402111a2:	006906        	j	4021134a <_ZN12Adafruit_GFX5writeEh+0x1d2>
402111a5:	1a0c32        	l8ui	a3, a12, 26
402111a8:	180c82        	l8ui	a8, a12, 24
402111ab:	939c      	beqz.n	a3, 402111c8 <_ZN12Adafruit_GFX5writeEh+0x50>
402111ad:	089c32        	l16si	a3, a12, 16
402111b0:	904880        	addx2	a4, a8, a8
402111b3:	904430        	addx2	a4, a4, a3
402111b6:	069c32        	l16si	a3, a12, 12
402111b9:	0b2437        	blt	a4, a3, 402111c8 <_ZN12Adafruit_GFX5writeEh+0x50>
402111bc:	085c22        	s16i	a2, a12, 16
402111bf:	091c22        	l16ui	a2, a12, 18
402111c2:	b02820        	addx8	a2, a8, a2
402111c5:	095c22        	s16i	a2, a12, 18
402111c8:	089c32        	l16si	a3, a12, 16
402111cb:	099c42        	l16si	a4, a12, 18
402111ce:	0a1c62        	l16ui	a6, a12, 20
402111d1:	0b1c72        	l16ui	a7, a12, 22
[... 121 lines cut ...]
40211321:	1133d0        	slli	a3, a3, 3
40211324:	400300        	ssr	a3
40211327:	913040        	srl	a3, a4
4021132a:	144060        	extui	a4, a6, 0, 2
4021132d:	c06640        	sub	a6, a6, a4
40211330:	0668      	l32i.n	a6, a6, 0
40211332:	1144d0        	slli	a4, a4, 3
40211335:	400400        	ssr	a4
40211338:	914060        	srl	a4, a6
4021133b:	744040        	extui	a4, a4, 0, 8
4021133e:	f8e416        	beqz	a4, 402112d0 <_ZN12Adafruit_GFX5writeEh+0x158>
40211341:	744030        	extui	a4, a3, 0, 8
40211344:	f1a456        	bnez	a4, 40211262 <_ZN12Adafruit_GFX5writeEh+0xea>
40211347:	ffe146        	j	402112d0 <_ZN12Adafruit_GFX5writeEh+0x158>
4021134a:	7108      	l32i.n	a0, a1, 28
4021134c:	120c      	movi.n	a2, 1
4021134e:	61c8      	l32i.n	a12, a1, 24
40211350:	51d8      	l32i.n	a13, a1, 20
40211352:	20c112        	addi	a1, a1, 32
40211355:	f00d      	ret.n
40211357:	71dc00        	excw
4021135a:	124010        	excw
```

all fonts seem to cause it...

fishix10, 2016-01-06:
Yup. Same problem here. Been spending today trying to figure that out. I tried tracing and it comes down to this line in `Adafruit_GFX.cpp`:

            uint8_t   w     = pgm_read_byte(&glyph->width),
                  h     = pgm_read_byte(&glyph->height);

The code runs up until that line. Using the default 5x7 font works.

PaintYourDragon, 2016-01-06:
Fixed in most recent commit by markwal.

closed 2016-01-06 by PaintYourDragon