adafruit/Adafruit-GFX-Library #69: Simple method for x5 faster font rendering

Bodmer, 2016-01-03:
Count horizontal foreground pixels for 2 to 5 times faster font rendering with ILI9341 display.
Particularly beneficial for larger and bold fonts.

Replace code in drawChar() as follows

#define FAST_TEXT // Enable faster drawing of glyphs
#ifdef FAST_TEXT
    uint16_t hpc = 0; // Horizontal foreground pixel count
    for(yy=0; yy<h; yy++) {
      for(xx=0; xx<w; xx++) {
        if(bit == 0) {
          bits = pgm_read_byte(&bitmap[bo++]);
          bit  = 0x80;
        }
        if(bits & bit) hpc++;
        else {
          if (hpc) {
            if(size == 1) {
              drawFastHLine(x+xo+xx-hpc, y+yo+yy, hpc, color);
            } else {
              fillRect(x+(xo16+xx-hpc)*size, y+(yo16+yy)*size, size*hpc, size, color);
            }
            hpc=0;
          }
        }
        bit >>= 1;
      }
      // Draw pixels for this line as we are about to increment yy
      if (hpc) {
        if(size == 1) {
          drawFastHLine(x+xo+xx-hpc, y+yo+yy, hpc, color);
        } else {
          fillRect(x+(xo16+xx-hpc)*size, y+(yo16+yy)*size, size*hpc, size, color);
        }
        hpc=0;
      }
    }
#else
    for(yy=0; yy<h; yy++) {
      for(xx=0; xx<w; xx++) {
        if(!(bit++ & 7)) {
          bits = pgm_read_byte(&bitmap[bo++]);
        }
        if(bits & 0x80) {
          if(size == 1) {
            drawPixel(x+xo+xx, y+yo+yy, color);
          } else {
            fillRect(x+(xo16+xx)*size, y+(yo16+yy)*size, size, size, color);
          }
        }
        bits <<= 1;
      }
    }
#endif

Bodmer, 2016-01-03:
I ran a test on the speed improvement suggested using an Arduino Mega + ILI9341 TTFT with SPI interface and otherwise "stock" Adafruit libraries.

To print "Hello World" in font FreeSerifBold24pt7b 4 times took 2.1 seconds (yawn).

After the mod it took 266ms. So that is a 7.7x speed improvement which in my view makes it a worthwhile tweak.

Bodmer, 2016-01-03:
Ah I see the multiply symbol asterisk get removed by the format interpreter in the pasted code but I am sure you would have spotted that!

Bodmer, 2016-01-03:
Try to get it right this time:
<
#define FAST_TEXT // Enable faster drawing of glyphs
#ifdef FAST_TEXT
    uint16_t hpc = 0; // Horizontal foreground pixel count
    for(yy=0; yy<h; yy++) {
      for(xx=0; xx<w; xx++) {
        if(bit == 0) {
          bits = pgm_read_byte(&bitmap[bo++]);
          bit  = 0x80;
        }
        if(bits & bit) hpc++;
        else {
          if (hpc) {
            if(size == 1) drawFastHLine(x+xo+xx-hpc, y+yo+yy, hpc, color);
            else fillRect(x+(xo16+xx-hpc)*size, y+(yo16+yy)*size, size*hpc, size, color);
            hpc=0;
          }
        }
        bit >>= 1;
      }
      // Draw pixels for this line as we are about to increment yy
      if (hpc) {
        if(size == 1) drawFastHLine(x+xo+xx-hpc, y+yo+yy, hpc, color);
        else fillRect(x+(xo16+xx-hpc)*size, y+(yo16+yy)*size, size*hpc, size, color);
        hpc=0;
      }
    }
#else
    for(yy=0; yy<h; yy++) {
      for(xx=0; xx<w; xx++) {
        if(!(bit++ & 7)) {
          bits = pgm_read_byte(&bitmap[bo++]);
        }
        if(bits & 0x80) {
          if(size == 1) drawPixel(x+xo+xx, y+yo+yy, color);
          else fillRect(x+(xo16+xx)*size, y+(yo16+yy)*size, size, size, color);
        }
//>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
      // Temporary code to test draw speed (and minimal flicker!) with background
      // Yes there are still display artifacts but that is not the point!
      else {
        if (size == 1) {
          drawPixel(x + xo + xx, y + yo + yy, bg);
        } else {
          fillRect(x + (xo16 + xx)*size, y + (yo16 + yy)*size, size, size, bg);
        }
      }
//>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
        bits <<= 1;
      }
    }
#endif
>

Bodmer, 2016-01-03:
Code as an attachment
[Code.txt](https://github.com/adafruit/Adafruit-GFX-Library/files/76689/Code.txt)
 with better formatting!

jalopezsuarez, 2017-07-30:
Great improvement! Is really fast! 
Please try to add to the Master Branch.

Bodmer, 2017-10-20:
Closed due to lack of interest :-(

closed 2017-10-20 by Bodmer