adafruit/Adafruit-GFX-Library #72: fontconvert errors

HausofLars, 2016-01-28:
Greetings,
I attempted to execute fontconvert but am running into problems. When I run 'sh makefonts.sh', the following error is received:

$ sh makefonts.sh 
makefonts.sh: 18: makefonts.sh: Syntax error: "(" unexpected

Please advise.
*
Linux Mint17 x64

microbuilder, 2016-01-28:
Please post support inquiries on the Adafruit support forum, thanks.  The support staff can't monitor Github issues and doesn't get the email alerts.

closed 2016-01-28 by microbuilder

pljakobs, 2016-01-28:
my first hunch would be: what shell are you using? I assume anything other than bash is not really likely anymore, but the error you're getting is related to bash syntax. Can you check your aliases pleas? 
Try this:
$ alias
$ which sh
$ file $(which sh)
and post the results

HausofLars, 2016-02-01:
Ok. Don't open issues in the issues section. Got it. Opened post in forum 4 days ago, no response. Open here, immediate response. :/

@pljakobs: 
The output from my console shell:
$ alias
alias grep='grep --colour=auto'
alias ll='ls -al'
alias ls='ls --color=auto'

$ which sh
/bin/sh

$ file $(which sh)
/bin/sh: symbolic link to `dash'

pljakobs, 2016-02-01:
I'm no mint user, but my understanding is that mint uses dash instead of bash (as your last post shows). I assume that "dash" cannot assign a list to a variable in the 
`var=(list of words)`
style.
(just verified that by installing dash and trying to run the script - got the same error)
I assume that bash is not installed on your system (you could verify that by running `which bash`) and thus the `#!/bin/bash` line in the script cannot invoke a working bash shell either.
My suggestion: install bash using the usual package manager for your distribution and then try running the makefont.sh script again. If it still fails try running `/bin/bash -e makefont.sh`.
You have to be aware that this script makes a set of very specific assumptions, namely that there is a font directory named _freefont_ on your desktop (path _~/Desktop/freefont_). If that is not the case, it will fail silently.

As for the "go to the forum" suggestion further up: I think this is a good idea in many cases, when there are many more users of a piece of software than there are devs. With the fonts extension to Adafruit_GFX being rather novel, and fontconvert not being something that most people will use, chances are, there are not many experienced users on the forum yet.
I wrote the initial version of the fontconvert program and have plans to extend it, which is why I'm monitoring this git repo, which is why I could respond quickly.

hope that helps.

pj

HausofLars, 2016-02-01:
Thank you for your reply. I saw that as well but I did check and found that bash is installed when I checked the following:

$ which bash
/bin/bash

I did notice that the makefonts.sh script called for bash:
#!/bin/bash
 
But didn't realize another situation. The fontconvert.c had to be compiled :| A grave oversight but didn't see any instruction on the make. I had wiped the GFX folder and installed a fresh one. This time, I was able to run sudo make makefonts with no errors but I am unable to locate the fontconvert executable and have replied to an existing forum post for this:
https://forums.adafruit.com/viewtopic.php?f=47&t=87433&p=448230#p448230

HausofLars, 2016-02-01:
Also, Thank you for your time and patience. It is greatly appreciated and yes, I do agree that the forum will be the better path to take. Sorry @microbuilder, I didn't mean to snarky. I know you were trying to help. :}

pljakobs, 2016-02-01:
@HausofLars I have replied to the forum post - for the benefit of the others, too.