adafruit/Adafruit-GFX-Library #78: Can't compile fontconvert in Mint

alidaf, 2016-03-16:
There was an issue with the folders for freetype but I can't get past the not found error for ftglyph.h

pljakobs, 2016-03-16:
Have you installed the devel packages for freetype2? I think that should fix it

alidaf, 2016-03-17:
Yes, I have libfreetype6-dev installed but no cigar.

I have designed a set of bitmap font glyphs but can't find any way of converting them into Adafruit-gfx format. I've tried over 20 packages now and this is the only one that has that as an output. If I can get this to work then I can convert my glyphs into ttf and thence to Adafruit-gfx, or use the similar Glass ttf. There are a few different font formats floating around for Arduino et al but this one makes a bit more sense to me, especially in keeping it compact. It's also the most unique.

I've designed them in a libreoffice spreadsheet but I've been procrastinating writing a macro to do the conversion because I'd have thought that there was something out there already. I may just have to bite the bullet.

alidaf, 2016-03-17:
I have it working. I was trying to compile it directly when all I needed to do was make!

closed 2016-03-17 by alidaf

alidaf, 2016-03-18:
Hi
The converter didn't work well with my font (it has artifical scanlines so has to be specific sizes to avoid anti-aliasing - I'm trying to recreate a VT220 terminal look) so I decided to write a program to convert my spreadsheet output to the Adafruit-gfx format. I have managed to create the packed code but I don't quite fully understand what the xAdvance, xOffset & yOffset vars are.

If I have as an example, the char 'y' (8x20). The baseline is row 12 so the tail hangs below row 12 and the u part sits above. The uppermost left bound for the packed glyph is at row 4, col 0 and the lowermost right bound is at row 18, col 7. 

Is the cursor at the baseline (row 12), the bottom of the unpacked glyph (row 19), or below the lowermost pixel (also row 19 for this example)?

Cursor is at the baseline:
xAdvance = width + some space = 20 + 1 = 21.
xOffset = 0.
yOffset = 4 - 12 = -8.

Cursor at bottom (and below):
xAdvance = width + some space = 20 + 1 = 21.
xOffset = 0.
yOffset = 0 - 19 = -19.

Are any of these correct?

I can't test this yet as I only have an SSD1302 until I can get the Adafruit VFD. I was planning to have a stab at the SSD1322 driver myself in the meantime.

Regards