adafruit/Adafruit-GFX-Library #87: Why are methods such as drawChar() not virtual?

jayeye, 2016-06-06:
Is it to reduce the memory footprint on memory-starved cores? I have several good reasons to override some of them, in particular, if I'm using a monospaced font, it makes sense to be able to override drawChar() so it first clears the off-pixels of the bounding box of the largest character in the font, rather than blink the entire string. 

Shall I send a pull request?

kyle-emmerich, 2016-12-07:
Virtual functions are pretty costly even when you're writing code for high-end x86-64 CPUs. You definitely don't want that kind of overhead on a microcontroller.

In fact, on a microcontroller, I would want the smallest possible library footprint. This one is already fairly large; coupled with the driver for the display itself, I'm using about 35% of my 32K!

If you want some specific functionality, you might just want to fork it and implement it separately.

mikeysklar, 2026-11-08:
Thanks for the question. `write()` is virtual, so a subclass can change how each character is drawn without touching GFX. For clearing behind custom-font text, the README covers it: clear the area with `fillRect()`, or to avoid flicker draw into a `GFXcanvas1` and copy it to the screen with `drawBitmap()` using a background color. Closing this one.

closed 2026-11-08 by mikeysklar