adafruit/Adafruit-GFX-Library #88: ft2build.h  file is missing

HP-Sparks, 2016-06-21:
Hi,
I'm giving a try to a simple Arduino-based Oscilloscope project, the " fontconvert.c " file refers to a " ft2build.h " file to be included. (see line 23: `#include <ft2build.h>`)

Where can we find this file if Not present amongst the already present libraries on the computer ?

Thanks in advance.

Best regards,
  HP_

pljakobs, 2016-06-21:
Hi HP_,

what platform are you building on?
you'll need the freetype 2 development packages under Linux / cygwin.
I have tested it on Fedora 21, SuSE 13.x, Mint and cygwin.

pj

HP-Sparks, 2016-06-23:
Hi @pljakobs ,
Sorry, my bad... I was so focus on my work and the parallel hunts for data, files and information that I forgot to share this information... :-/

Well, I'm using the Arduino IDE 1.6.9 on a Linux Ubuntu-based distribution system.

After installing the latest stable IDE from the official Web site [(http://arduino.cc),](http://arduino.cc) I allowed all the automatic updates requests I faced for the bundled and newly available libraries.

Then, I also did a search on both Internet and my system, it seems that this file disappeared for many people for some reason(s). (copyright/licensing issue ?? trying to guess...)

Best regards,
HP_

HP-Sparks, 2016-06-23:
Hi @pljakobs,

Sorry, my bad... I was so focus on my work and the parallel hunts for data, files and information that I forgot to share this information... :-/
Well, I'm using the Arduino IDE 1.6.9 on a Linux Ubuntu-based distribution system.

Now, going to search for _Font Type 2_ development packages...
But, yesterday, searching on my system (using `me@netbook / # find / ft2build.h
`) or on Internet, I found nothing except many people facing the same issue for all kind of projects developed using all kind of programming languages...
So, I'm wondering if this file has been removed due to copyright/licensing issue... ?? (trying to guess...)

Thank you for the clue ! :-)

Best regards,
HP_

**P.S.**
Just in case there would be a question on the Arduino IDE:
 - After installing the latest stable IDE from the official Web site [(http://arduino.cc),](http://arduino.cc) I allowed all the automatic updates requests I faced for the bundled and newly available libraries.
 - No magic from this too. ;-)

HP-Sparks, 2016-06-23:
Well, I found something from there: http://download.savannah.gnu.org/releases/freetype/freetype-2.6.tar.gz (latest version at the moment)
Just wondering if this will work fine with code for Arduino... ??
(Now, I have to try this when time permits, I had to switch to alternative solution for my project needs...)

pljakobs, 2016-06-23:
Ah, couple of things:  the font converter is not an arduino sketch, it cannot be compiled with the arduino IDE but has to be compiled with the gcc that comes with your Linux distro. So you have to install the current development environment for your platform plus the freetype2 system along with it's development packages. You would do that through the package manager provided by your Linux distro (for example aptitude on many Debian based systems, or apt-cache / apt-get at the command line). Those tools have a search function that allows you to find the correct packages and meta packages. What you want is something along the lines of "C development", and "freetype2" - for the latter, there will be an option to also install the devel package. 
Those packages are all free software, so there's no copyright issue with them.

pj

CraigLindley, 2016-07-23:
I'm having the same problem on OSX when I am compiling for the esp8266. I've downloaded the newest version of the Arduino IDE, installed the newest ESP8266/Arduino code and included adafruit GFX in my project. Compilation always fails because of the missing ft2build.h file

bcr, 2016-08-18:
I can submit a patch for this if you want it. It sounds like people are incorporating the entire GFX repo into their projects, not using the Arduino library management. If you incorporate the entire GFX repo into your sketch directory, the Arduino IDE will try to build `fontconvert.c`

I think a simple fix would be to add `#ifndef ARDUINO` at the top and `#endif /* !ARDUINO */` at the end. This fixes a separate issue that I'm having with PlatformIO's library management (which I think is a PlatformIO bug -- I think they're building stuff that isn't in the `src` or the root directory).

savvycreativeadventurer, 2017-05-13:
Hi BRC 

where should we add those codes please?

Johnvan, 2017-11-29:
I am having the same problem with ft2build.h in Windows 10 as the IDE keeps trying to load it from the Programs(x64) directory (cores etc.) and when I do a search for ft2build.h it does not find it in the cores directory setup. **It is in the Sketches directory** (it is not visible in this directory but the search says it's there) but apparently the IDE only looks for the hardware cores directory. I copy/paste it into the hardware/cores directory then search for it there but the search does not find it either. Why is that

pljakobs, 2017-11-29:
@Johnvan please see my comment above, you cannot build fontconvert with the Arduino ide since it's not an Arduino Sketch, it's a program that has to be run natively on your platform (Linux, MacOS or Windows). Since I don't do any coding on Windows, I have no compiler that can create Windows code (would there even be free ones?).
But there has been work done to make it compile on Windows, please look around in the actual repo and the issues here.

pj

Johnvan, 2017-11-30:
Hi Peter,

Thanks for your quick and accurate response. I had so much trouble with this that I even reloaded the Arduino IDE and also got into the Web version IDE.

This has made me tired of all this mucking around so I will have to throw this 2.8TFT TJCTM24028-SPI out. This is getting worse and worse with non-Arduino modules and I am about to delete all my Arduino apps and look into something else.

Thanks anyway.

John van der Linden.

pljakobs, 2017-11-30:
Hi John,

did you see https://github.com/adafruit/Adafruit-GFX-Library/blob/master/fontconvert/fontconvert_win.md ? I think there should be a rather straightforward path to building fontconvert on Windows... 
But: what do you need it for, anyway? Do you have a specific requirement for a font that cannot be replaced with one of those already available in the fonts subdirectory?
I'm using the GFX_library fonts in many places with no issues and the only reason to use the fontconvert is when I do want a different optics (like a faux 7 segment display or so).
It makes me a wee bit unhappy to see you drop a project due to this, so if I can help, let me know (ideally by e-mail, address is in my profile)

kind regards

pj

aaseagrass, 2018-03-14:
Hello! I have an error while trying to install the matplotlib library in python 3.4, I found out that it needs ft2build.h which I already downloaded, but I can't find command to install it. Can you help me? What command to us to install ft2build.h? I'm using windows 10 thank you! I'm a graduating student and we badly need this in our thesis.

pljakobs, 2018-03-16:
@aaseagrass you are posting this question to the github repo for Adafruit-gfx, this has nothing to do with python. Please check out the [freetype](https://www.freetype.org/) website on how to install the freetype develop packages on your platform.

Waged, 2018-10-01:
hi, I've faced the same problem to create the .h file from .ttf file , I'm using MacOS  
after trials i made  this procedures and it works 👍 
1- install brew: link :http://macappstore.org/freetype/
2-install freetype2 link : https://opensource.ncsa.illinois.edu/confluence/pages/viewpage.action?pageId=61604031 
3- run command : make fontconvert
4- after downloading the .ttf file  and putting it inside the fonconvert folder run command:
./fontconvert Orgy.ttf  18  > Orgy18.h

theintel, 2018-10-24:
I've referenced this issue, committed a change and opened a pull request. Its already been merged. Thanks to @bcr for his quick fix.

This issue is resolved now.

mikeysklar, 2026-10-16:
Resolved with PR #463, which is now merged. Thanks for reporting this!

closed 2026-10-16 by mikeysklar