adafruit/Adafruit-GFX-Library #90: getTextBounds should have String as param as well

xmenxwk, 2016-08-23:
How to resolve this ?

        String someText = "SomeText";//generated by a method
	int16_t x1, y1;
	uint16_t w1, h1; 
	tft.getTextBounds(someText , 0, 0, &x1, &y1, &w1, &h1);//error here

Tried using c_str();, doesn't help either.

amitkumarsharma008, 2016-09-18:
int str_len = someText .length() + 1; 
    char char_array[str_len];
    someText .toCharArray(char_array, str_len);
    lcd.getTextBounds(char_array, 1, 56, &upperleftX, &upperleftY, &width, &height);

minida28, 2018-07-30:
Usage of `String` is considered as a bad practice in embedded devices with limited memory, so it should be avoided as hard as possible. Maybe @ladyada will not implement your request due to that reason.

Here is a good reading:

[https://hackingmajenkoblog.wordpress.com/2016/02/04/the-evils-of-arduino-strings/](https://hackingmajenkoblog.wordpress.com/2016/02/04/the-evils-of-arduino-strings/)

ladyada, 2018-07-30:
hiya - yes we strongly prefer no Strings plz :)

vortigont, 2018-08-03:
I'm not promoting Strings here. Actually using this lib it is possible to .print() a String to the LCD,
but at the same time you can't find it's size with .getTextBounds(). This doesn't look consistent at all when some methods works on the object and some doesn't. If we do not like Strings in one place than why use it in others? Besides, the fix that I offered doesn't use Strings manipulation at all, it just casts it as a const char pointer, enough to calculate WxH on the screen.

ladyada, 2018-08-03:
ooh right. its been a while since we wrote that code. @vortigont can you rebase since we've made a ton of changes and it isnt a clean merge?

dragondaud, 2018-08-05:
I manually patched to use the String overload, and would like to see that merged. The argument against Strings is fine on tiny microcontrollers, but the $2 ESP8266 has plenty of RAM, so the point is moot.

vortigont, 2018-08-06:
@ladyada  done. pls, review.
thanks for looking into this.

ladyada, 2018-08-12:
https://github.com/adafruit/Adafruit-GFX-Library/pull/130#issuecomment-412334177

closed 2018-08-12 by ladyada