adafruit/Adafruit-GFX-Library #92: Adafruit GFX Library compiling problems with eclipse?

Sniperblue, 2016-10-02:
I can't compile the Adafruit GFX Library with eclipse! Instead I get always the following problem report from compiler? Have got anyone a solution?

Description	Resource	Path	Location	Type
'bitmap' undeclared (first use in this function)	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 159	C/C++ Problem
'face' undeclared (first use in this function)	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 119	C/C++ Problem
'FT_BitmapGlyphRec' undeclared (first use in this function)	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 160	C/C++ Problem
'FT_LOAD_TARGET_MONO' undeclared (first use in this function)	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 140	C/C++ Problem
'FT_RENDER_MODE_MONO' undeclared (first use in this function)	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 147	C/C++ Problem
'g' undeclared (first use in this function)	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 160	C/C++ Problem
'glyph' undeclared (first use in this function)	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 153	C/C++ Problem
'library' undeclared (first use in this function)	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 115	C/C++ Problem
expected expression before ')' token	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 160	C/C++ Problem
Field 'advance' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 173	Semantic Error
Field 'bitmap' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 159	Semantic Error
Field 'buffer' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 181	Semantic Error
Field 'glyph' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 146	Semantic Error
Field 'glyph' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 153	Semantic Error
Field 'glyph' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 159	Semantic Error
Field 'glyph' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 173	Semantic Error
Field 'height' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 226	Semantic Error
Field 'left' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 174	Semantic Error
Field 'metrics' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 226	Semantic Error
Field 'pitch' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 182	Semantic Error
Field 'rows' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 172	Semantic Error
Field 'rows' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 177	Semantic Error
Field 'rows' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 187	Semantic Error
Field 'rows' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 192	Semantic Error
Field 'size' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 226	Semantic Error
Field 'top' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 175	Semantic Error
Field 'width' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 171	Semantic Error
Field 'width' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 178	Semantic Error
Field 'width' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 187	Semantic Error
Field 'width' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 192	Semantic Error
Field 'x' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 173	Semantic Error
make: *** [Adafruit-GFX/fontconvert/fontconvert.o] Error 1	UNO_OLED128x64_LibTest		 	C/C++ Problem
Symbol 'FT_LOAD_TARGET_MONO' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 140	Semantic Error
Symbol 'FT_RENDER_MODE_MONO' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 147	Semantic Error
Type 'FT_Bitmap' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 56	Semantic Error
Type 'FT_BitmapGlyphRec' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 57	Semantic Error
Type 'FT_BitmapGlyphRec' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 160	Semantic Error
Type 'FT_Face' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 54	Semantic Error
Type 'FT_Glyph' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 55	Semantic Error
Type 'FT_Library' could not be resolved	fontconvert.c	/UNO_OLED128x64_LibTest/Adafruit-GFX/fontconvert	line 53	Semantic Error

pljakobs, 2016-10-30:
it would seem that you don't have the relevant includes for freetype. Did you download / install the freetype library and it's devel package?

Sniperblue, 2016-11-02:
No, I was not aware that this is neccessary to run the ssd1306_128x64_i2c example I thought all neccessary files are already include in the GFX Lib. Are there any howto for including the neccessary files?

tonitotti, 2019-12-01:
hello guys, I am having a problem of including Adafruit_GPS.h library in my interfacing project with Cyclone v Eclipse, I donot know how to customize the library as there is software serial in it and i donot know how to make GLOBAL library. Any help appreciated.

mikeysklar, 2026-10-21:
The errors come from fontconvert, a desktop tool, not the Arduino library.
fontconvert needs FreeType, and no other file in the library does.
Leave the fontconvert folder out of your Eclipse project to fix the errors.

closed 2026-10-21 by mikeysklar