adafruit/Adafruit-GFX-Library #93: A guide of how to use fontconvert.c

johnbacas, 2016-10-27:
Hi !
I want to use the fontconvert.c file to generate a new font library from a .ttf file
I have installed MinGW and MSYS on my computer (running windows10) and also downloaded a freetype library (a folder with name : "freetype-2.3.5.1-lib" which contains  "include", "lib" and "manifest" folders)
I have included C:MinGW\bin & C:MinGW\MSYS\1.0\bin in my PATH variable.
What should I do next ??

I run the command "gcc fontconvert.c" but I got an error message that it cannot find "ft2build.h"
Should I copy the library folder somewhere in the C:\MinGW directory ??

Thanks

pljakobs, 2016-10-27:
going by the path names, I suspect you're trying to build it in some kind of Microsoft Windows compiler?
It was originally created for gcc on either Linux, MacOS or, if it should be Windows, in the Cygwin Environment. I have admit that I have never done any c-development in Windows and thus can't really help here. My only suggestion would be installing cygwin from http://cygwin.org/ with the relevant libraries, but unless you have at least basic experience with unix-like environments, the learning curve might be a little steep here.

johnbacas, 2016-10-27:
At freetype.org they suggest using MinGw instead of Cygwin...(last update 2013)
https://www.freetype.org/download.html

I'm not familiar with UNIX-like environments so I'll call it a day...

thanks anyway for the responce

pljakobs, 2016-10-27:
I'm sorry I can't be of more help. In theory, it should easily compile an any standard C99 environment where the freetype development libraries are available. It's just that I don't know how to do that on Windows.
I can only suggest to follow https://gcc.gnu.org/wiki/WindowsBuilding and the relevant Freetype docs and though shall prevail...

johnbacas, 2016-10-28:
Hi again !
In the "makefile" there are 3 variables: CC , CFLAGS and LIBS
the first refers to the compiler (gcc) , the second is the path to find the Freetype2 directory
what is the third  (-lfreetype) ??

pljakobs, 2016-10-28:
The cflags swich on all warnings (-Wall) and point to  the include directories (where the .h files are found -I/usr/local/include/freetype2 -I/usr/include/freetype2 -I/usr/include)
The -lfreetype tells the toolchain to link the freetype library.

microbuilder, 2016-10-28:
That describes the libraries to include when linking the custom code (freetype in this case). I'm afraid explaining the ins and outs of makefiles, GCC compiler options, etc., really isn't something that will ever be solved in a Github issue, though. :(

If you're really serious about learning how to use GCC, I would suggest starting with a simple hello world type project and makefile to get things setup on your system, etc.

microbuilder, 2016-10-28:
Closing this issue, but just doing a quick search I came across this for example:

- http://www.network-theory.co.uk/docs/gccintro/index.html
Compiling a simple program (to test the compiler etc.): http://www.network-theory.co.uk/docs/gccintro/gccintro_9.html
Working with a Makefile: http://www.network-theory.co.uk/docs/gccintro/gccintro_16.html
Libraries (which relates to your question above): http://www.network-theory.co.uk/docs/gccintro/gccintro_17.html

Hope this helps.

closed 2016-10-28 by microbuilder

johnbacas, 2016-10-29:
My contribution to human kind !!
[GRR9pt7b.zip](https://github.com/adafruit/Adafruit-GFX-Library/files/559825/GRR9pt7b.zip)

display.println("EUCARISTW")

For the "poor" windows users, I would like to share my experience with a guide of how to use the fontconvert.c .
What is the best way to do so in Github ??

microbuilder, 2016-10-29:
You could perhaps create a text file in markdown format (`windowsreadme.md`?), and submit that as a pull request? You create a pull request by forking this repo to your local account, making the changes and then pushing them to your account, and then you create a Pull Request in the Github IDE which will send it back to the parent repo and we can review it.

microbuilder, 2016-10-29:
If you're new to Markdown, here is a quick guide: https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet

It's the main display formatting scheme use on Github, so it's good to know the basics about it anyway. You can use a text editor like Atom to preview Github as well (Markdown preview is included by default as a plugin), though there are dedicated markdown editors out there as well.