adafruit/Adafruit-GFX-Library #98: Hey, i have an arduino 101 and iam trying to run it with the adafruit st7735 tft and iam getting some errors wjile compiling .

tanann, 2016-11-09:


tanann, 2016-11-09:
Arduino: 1.6.12 (Windows 10), Board: "Arduino/Genuino 101"

C:\Users\tanma\OneDrive\Documents\Arduino\libraries\Adafruit-ST7735-Library-master\Adafruit_ST7735.cpp: In constructor 'Adafruit_ST7735::Adafruit_ST7735(int8_t, int8_t, int8_t, int8_t, int8_t)':

C:\Users\tanma\OneDrive\Documents\Arduino\libraries\Adafruit-ST7735-Library-master\Adafruit_ST7735.cpp:47:3: error: '_cs' was not declared in this scope

   _cs   = cs;

   ^

C:\Users\tanma\OneDrive\Documents\Arduino\libraries\Adafruit-ST7735-Library-master\Adafruit_ST7735.cpp:48:3: error: '_rs' was not declared in this scope

   _rs   = rs;

   ^

C:\Users\tanma\OneDrive\Documents\Arduino\libraries\Adafruit-ST7735-Library-master\Adafruit_ST7735.cpp:49:3: error: '_sid' was not declared in this scope

   _sid  = sid;

   ^

C:\Users\tanma\OneDrive\Documents\Arduino\libraries\Adafruit-ST7735-Library-master\Adafruit_ST7735.cpp:50:3: error: '_sclk' was not declared in this scope

   _sclk = sclk;

   ^

C:\Users\tanma\OneDrive\Documents\Arduino\libraries\Adafruit-ST7735-Library-master\Adafruit_ST7735.cpp:51:3: error: '_rst' was not declared in this scope

   _rst  = rst;

   ^

C:\Users\tanma\OneDrive\Documents\Arduino\libraries\Adafruit-ST7735-Library-master\Adafruit_ST7735.cpp: In constructor 'Adafruit_ST7735::Adafruit_ST7735(int8_t, int8_t, int8_t)':

C:\Users\tanma\OneDrive\Documents\Arduino\libraries\Adafruit-ST7735-Library-master\Adafruit_ST7735.cpp:60:3: error: '_cs' was not declared in this scope

   _cs   = cs;

   ^

C:\Users\tanma\OneDrive\Documents\Arduino\libraries\Adafruit-ST7735-Library-master\Adafruit_ST7735.cpp:61:3: error: '_rs' was not declared in this scope

   _rs   = rs;

   ^

C:\Users\tanma\OneDrive\Documents\Arduino\libraries\Adafruit-ST7735-Library-master\Adafruit_ST7735.cpp:62:3: error: '_rst' was not declared in this scope

   _rst  = rst;

   ^

C:\Users\tanma\OneDrive\Documents\Arduino\libraries\Adafruit-ST7735-Library-master\Adafruit_ST7735.cpp:64:3: error: '_sid' was not declared in this scope

   _sid  = _sclk = 0;

   ^

C:\Users\tanma\OneDrive\Documents\Arduino\libraries\Adafruit-ST7735-Library-master\Adafruit_ST7735.cpp:64:11: error: '_sclk' was not declared in this scope

   _sid  = _sclk = 0;

           ^

C:\Users\tanma\OneDrive\Documents\Arduino\libraries\Adafruit-ST7735-Library-master\Adafruit_ST7735.cpp: In member function 'void Adafruit_ST7735::spiwrite(uint8_t)':

C:\Users\tanma\OneDrive\Documents\Arduino\libraries\Adafruit-ST7735-Library-master\Adafruit_ST7735.cpp:93:20: error: 'dataport' was not declared in this scope

[... 332 lines cut ...]
   *rsport |=  rspinmask;

               ^

C:\Users\tanma\OneDrive\Documents\Arduino\libraries\Adafruit-ST7735-Library-master\Adafruit_ST7735.cpp:530:4: error: 'csport' was not declared in this scope

   *csport &= ~cspinmask;

    ^

C:\Users\tanma\OneDrive\Documents\Arduino\libraries\Adafruit-ST7735-Library-master\Adafruit_ST7735.cpp:530:15: error: 'cspinmask' was not declared in this scope

   *csport &= ~cspinmask;

               ^

Multiple libraries were found for "Adafruit_GFX.h"
 Used: C:\Program Files (x86)\Arduino\libraries\Adafruit_GFX
 Not used: C:\Users\tanma\OneDrive\Documents\Arduino\libraries\Adafruit-GFX-Library-master
exit status 1
Error compiling for board Arduino/Genuino 101.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

pljakobs, 2016-11-09:
```
Used: C:\Program Files (x86)\Arduino\libraries\Adafruit_GFX
Not used: C:\Users\tanma\OneDrive\Documents\Arduino\libraries\Adafruit-GFX-Library-master
```
this tells you your IDE is not in fact using the gihub version of Adafruit_GFX - you may want to check what version you have installed

you can include the lib you want by providing the full path.

The errors you're getting further up may or may not follow from that.

It seems highly unlikely that this is a bug in Adafruit_GFX so you'd probably be best off asking that question on the more user-focussed [adafruit forums](https://forums.adafruit.com/index.php)

pj

microbuilder, 2016-11-09:
As mentioned above, please raise this on the Adafruit support forums. It isn't a library issue. Support happens there, github is only for pull requests and bugs with the library. Thanks!

closed 2016-11-09 by microbuilder