adafruit/Adafruit-GFX-Library #103: fontconvert: created fonts look terrible

elFallino, 2017-01-05:
Hi,

to have some german umlaut characters I used the fontconvert tool (on windows, minGW, freefonts2.7 following fontconvert_win.md) to convert FreeSans.

How ever, results do work but look very ugly. They look as if their width does not match and therefore are jolted (or compressed/forged - not sure how to translate).
When I try to convert the same TTF with default settings (witch *should* create the same *.h file as shipped) the result is a bit different and looks ugly, too.

Both files differ in size:

Creating a 9pt header it results in an reported estimated size "// Approx. 1801 bytes" while the original file reports "// Approx. 1822 bytes"

pljakobs, 2017-01-07:
Hi elFallino, my original work was to exactly enable international (well, in my case, German) character sets which requires larger font tables. When this was picked up by Adafruit, they decided to settle for 7 bit characters for now, mainly because 8 bit would require up to twice the memory space plus character mapping - if we do it, we don't just want to enable German characters, but ideally all iso8859 character tables. I have not yet had the time to come up with a good idea on how to implement this, but I certainly have the need, so while I won't give an ETA, it will arrive at some point.
regards,

pj

elFallino, 2017-01-07:
Great. Let me know if I can assist you somehow. (testing, coding, share ideas).
You can contact me in German if you like to ;)

pljakobs, 2017-01-07:
Let's keep this in English in case someone else wants to chime in

Here's my idea: 
ISO8859 encoding is identical for the lower 7 bits, so that can stay the same, above that, from 0xA0 to 0xFF is where the different encodings differ. See https://de.wikipedia.org/wiki/ISO_8859#Tabelle_der_regionalen_Sonderzeichen
To enable GFXLibrary to use 8 bit fonts, we need to 

- modify fontconvert to encode 8-bit fonts for different encodings using utf->iso8859 mapping
- modify GFXLibrary to use those fonts by mapping the incoming utf8 strings to is8859 characters

The latter is the rather more difficult bit, since this mapping is, for most iso8859 mappings, not mathematical, so it would require rather large conditional structures.
A while ago, I wrote a simple script that uses iconv to generate c-structures for this mapping, but of course, it's impractical to have an array of 2^16 values of which all but 192 are NUL.

I think the write() function would be the right place to do the utf->iso8859 mapping.
Also, I think there should be a way of stating at compile time what iso mapping should be used, that could then a) include the correct fonts (if they are named according to a convention) and b) enable a utf->iso8859 function for the desired mapping. This way, we would not have to carry all the mapping code in the binary.

an initial discussion was [here ](https://github.com/adafruit/Adafruit-GFX-Library/issues/64)

So you see, my thought process is far from finished, other projects have side-dragged me, but I wouldn't mind moving ahead with this.

these are my initial iso8859-x to utf conversion tables:

`
/*
 * converting iso8859-1
 */
int iso8859-1[]={
0xc2a0, 0xc2a1, 0xc2a2, 0xc2a3, 0xc2a4, 0xc2a5, 0xc2a6, 0xc2a7
0xc2a8, 0xc2a9, 0xc2aa, 0xc2ab, 0xc2ac, 0xc2ad, 0xc2ae, 0xc2af
0xc2b0, 0xc2b1, 0xc2b2, 0xc2b3, 0xc2b4, 0xc2b5, 0xc2b6, 0xc2b7
0xc2b8, 0xc2b9, 0xc2ba, 0xc2bb, 0xc2bc, 0xc2bd, 0xc2be, 0xc2bf
0xc380, 0xc381, 0xc382, 0xc383, 0xc384, 0xc385, 0xc386, 0xc387
0xc388, 0xc389, 0xc38a, 0xc38b, 0xc38c, 0xc38d, 0xc38e, 0xc38f
0xc390, 0xc391, 0xc392, 0xc393, 0xc394, 0xc395, 0xc396, 0xc397
0xc398, 0xc399, 0xc39a, 0xc39b, 0xc39c, 0xc39d, 0xc39e, 0xc39f
0xc3a0, 0xc3a1, 0xc3a2, 0xc3a3, 0xc3a4, 0xc3a5, 0xc3a6, 0xc3a7
0xc3a8, 0xc3a9, 0xc3aa, 0xc3ab, 0xc3ac, 0xc3ad, 0xc3ae, 0xc3af
0xc3b0, 0xc3b1, 0xc3b2, 0xc3b3, 0xc3b4, 0xc3b5, 0xc3b6, 0xc3b7
0xc3b8, 0xc3b9, 0xc3ba, 0xc3bb, 0xc3bc, 0xc3bd, 0xc3be, 0xc3bf
};

/*
 * converting iso8859-2
 */
int iso8859-2[]={
0xc2a0, 0xc484, 0xcb98, 0xc581, 0xc2a4, 0xc4bd, 0xc59a, 0xc2a7
0xc2a8, 0xc5a0, 0xc59e, 0xc5a4, 0xc5b9, 0xc2ad, 0xc5bd, 0xc5bb
0xc2b0, 0xc485, 0xcb9b, 0xc582, 0xc2b4, 0xc4be, 0xc59b, 0xcb87
0xc2b8, 0xc5a1, 0xc59f, 0xc5a5, 0xc5ba, 0xcb9d, 0xc5be, 0xc5bc
0xc594, 0xc381, 0xc382, 0xc482, 0xc384, 0xc4b9, 0xc486, 0xc387
0xc48c, 0xc389, 0xc498, 0xc38b, 0xc49a, 0xc38d, 0xc38e, 0xc48e
0xc490, 0xc583, 0xc587, 0xc393, 0xc394, 0xc590, 0xc396, 0xc397
0xc598, 0xc5ae, 0xc39a, 0xc5b0, 0xc39c, 0xc39d, 0xc5a2, 0xc39f
0xc595, 0xc3a1, 0xc3a2, 0xc483, 0xc3a4, 0xc4ba, 0xc487, 0xc3a7
0xc48d, 0xc3a9, 0xc499, 0xc3ab, 0xc49b, 0xc3ad, 0xc3ae, 0xc48f
0xc491, 0xc584, 0xc588, 0xc3b3, 0xc3b4, 0xc591, 0xc3b6, 0xc3b7
0xc599, 0xc5af, 0xc3ba, 0xc5b1, 0xc3bc, 0xc3bd, 0xc5a3, 0xcb99
};

/*
 * converting iso8859-3
 */
int iso8859-3[]={
0xc2a0, 0xc4a6, 0xcb98, 0xc2a3, 0xc2a4, 0x0000, 0xc4a4, 0xc2a7
0xc2a8, 0xc4b0, 0xc59e, 0xc49e, 0xc4b4, 0xc2ad, 0x0000, 0xc5bb
0xc2b0, 0xc4a7, 0xc2b2, 0xc2b3, 0xc2b4, 0xc2b5, 0xc4a5, 0xc2b7
0xc2b8, 0xc4b1, 0xc59f, 0xc49f, 0xc4b5, 0xc2bd, 0x0000, 0xc5bc
0xc380, 0xc381, 0xc382, 0x0000, 0xc384, 0xc48a, 0xc488, 0xc387
0xc388, 0xc389, 0xc38a, 0xc38b, 0xc38c, 0xc38d, 0xc38e, 0xc38f
0x0000, 0xc391, 0xc392, 0xc393, 0xc394, 0xc4a0, 0xc396, 0xc397
0xc49c, 0xc399, 0xc39a, 0xc39b, 0xc39c, 0xc5ac, 0xc59c, 0xc39f
[... 197 lines cut ...]
0xc398, 0xc399, 0xc39a, 0xc39b, 0xc39c, 0xc39d, 0xc39e, 0xc39f
0xc3a0, 0xc3a1, 0xc3a2, 0xc3a3, 0xc3a4, 0xc3a5, 0xc3a6, 0xc3a7
0xc3a8, 0xc3a9, 0xc3aa, 0xc3ab, 0xc3ac, 0xc3ad, 0xc3ae, 0xc3af
0xc3b0, 0xc3b1, 0xc3b2, 0xc3b3, 0xc3b4, 0xc3b5, 0xc3b6, 0xc3b7
0xc3b8, 0xc3b9, 0xc3ba, 0xc3bb, 0xc3bc, 0xc3bd, 0xc3be, 0xc3bf
};

/*
 * converting iso8859-16
 */
int iso8859-16[]={
0xc2a0, 0xc484, 0xc485, 0xc581, 0xe282ac, 0xe2809e, 0xc5a0, 0xc2a7
0xc5a1, 0xc2a9, 0xc898, 0xc2ab, 0xc5b9, 0xc2ad, 0xc5ba, 0xc5bb
0xc2b0, 0xc2b1, 0xc48c, 0xc582, 0xc5bd, 0xe2809d, 0xc2b6, 0xc2b7
0xc5be, 0xc48d, 0xc899, 0xc2bb, 0xc592, 0xc593, 0xc5b8, 0xc5bc
0xc380, 0xc381, 0xc382, 0xc482, 0xc384, 0xc486, 0xc386, 0xc387
0xc388, 0xc389, 0xc38a, 0xc38b, 0xc38c, 0xc38d, 0xc38e, 0xc38f
0xc490, 0xc583, 0xc392, 0xc393, 0xc394, 0xc590, 0xc396, 0xc59a
0xc5b0, 0xc399, 0xc39a, 0xc39b, 0xc39c, 0xc498, 0xc89a, 0xc39f
0xc3a0, 0xc3a1, 0xc3a2, 0xc483, 0xc3a4, 0xc487, 0xc3a6, 0xc3a7
0xc3a8, 0xc3a9, 0xc3aa, 0xc3ab, 0xc3ac, 0xc3ad, 0xc3ae, 0xc3af
0xc491, 0xc584, 0xc3b2, 0xc3b3, 0xc3b4, 0xc591, 0xc3b6, 0xc59b
0xc5b1, 0xc3b9, 0xc3ba, 0xc3bb, 0xc3bc, 0xc499, 0xc89b, 0xc3bf
};
`

elFallino, 2017-01-07:
Sure, everything in GitHub should stay in english.
I'll have a look later - I guess there are some things I have to understand. The ideas I had seem to be thought too easy :)
But one thing is sure, there need to be a way to leave out unneeded chars. If I need German special chars I do not need all those others that are in between. They take up too much space even if one uses a rather capable system (I am developing on ESP32 atm).

pljakobs, 2017-01-07:
well yes, of course we can make half-baked solutions but that would mean to create [Mojibake](https://en.wikipedia.org/wiki/Mojibake)

elFallino, 2017-01-07:
Now re-reading your post after my children are gone to bed I guess I understand what you are writing about (I'm sorry :) ).
So this would not fix my primary problem (incorrect converted fonts: everything is there but it seems that some pixels are missing or are there in addition ... like a bitmap scaled by a odd factor) but is indeed something worth to be done.

I shipped around the utf8-> ISO problematic by using escape sequences in code.
This is something you can do when just want to use some few special chars but surely is a pain in the ass with some other languages.

I am sure I do not completely understand your thoughts on that because I haven't spend a single brain-cell on that issue before.
But I still think that there should be an option to the converter itself to specify which glyphs are converted because in most cases (at least in western europe languages) you do not need more that one third of the 255 available glyphs and so they do not need to be stored into progmem.

This could be done by having the user specifying the glyphs that are wanted within a separate (e.g. .txt) file or passing as an additional parameter instead of defining a range of glyphs.

elFallino, 2017-01-08:
Here you can see a picture. It compares an original shipped one with a self-converted one.

https://postimg.org/image/6ml1xzfdp/79bfeb7a/

pljakobs, 2017-01-09:
ah, now I get it, I was under the impression that *only* the umlauts were being butchered. This looks much like rounding errors / dithering errors / pixel misalignment. I had seen that very early on when working with greyscale font rendering and then bad dithering. Did you use the fontconvert tool from this repo? It looks a lot like my old initial version...

elFallino, 2017-01-09:
I used the one that was shipped with Adafruit_GFX library. I am pretty sure I downloaded it from this git repo. But wouldn't swear so I'll check later to be sure.

pljakobs, 2017-01-11:
And you said you're using it on Windows?
I have never myself tested that port - I only work on Linux platforms. Can you try some other font and see if the results are the same?

elFallino, 2017-01-11:
Yes, I am using it on Windows. Even if I prefer Linux, too.
I'll try another font.

elFallino, 2017-01-14:
Other fonts look terrible, too. I tried FreeMono and FreeSerif.

pljakobs, 2017-01-17:
strange. I'm rather consumed by work and other projects right now, but I think I might have to try and recreate this.
I know it's not much help, but others have not reported the same issue, so I wonder what the reason might be. You might want to try and find the issue that resulted in the eventual port over to Windows and ask the fine chap who did that if he has seen similar problems?
Sorry that I cannot be of more help right now.
pj

PaintYourDragon, 2017-01-17:
The pre-converted example fonts included with GFX were produced on a system with FreeType 2.6.0 installed. Release notes for later versions of FreeType (2.6.4 and 2.7) mention some changes in the default hinting behavior...this may or may not be what you're seeing. Release notes also mention that the "classic" behavior can be configured at compile-time or through the FREETYPE_PROPERTIES environment variable.

elFallino, 2017-01-17:
Thanks. Good hint.

I tried 
`$ env FREETYPE_PROPERTIES=freetype fontconvert.exe FreeSans.ttf 9 32 252 > FreeSans9pt8b_2.h` (should be the new hinting function)
and
`$ env FREETYPE_PROPERTIES=adobe fontconvert.exe FreeSans.ttf 9 32 252 > FreeSans9pt8b_2.h` (should be the old hinting function)
but both results do not differ (using a diff-tool) and they also do not differ from what I have produced before (the one that looks so bad).

PaintYourDragon, 2017-01-17:
Suggest maybe installing 2.6, see if any change...can always re-install latest over it if needed.

PaintYourDragon, 2017-01-17:
Looks like properties syntax for hinting might be:
FREETYPE_PROPERTIES=cff:hinting-engine=adobe
Hard to say for sure. Documentation's a little sparse there and the source is like a billion things to dig through. Well, if earlier release works, maybe that's just the easier thing to do for now.

pljakobs, 2017-01-18:
thanks @PaintYourDragon, I was unaware of that change and, in fact, hadn't used fontconvert in a while.

mertaysoy, 2017-01-29:
I have exactly the same issue. Tried list:

freetype 2.7.1 and freetype 2.6.0 
Windows and Mac OS X
"FREETYPE_PROPERTIES=cff:hinting-engine=adobe" and "FREETYPE_PROPERTIES=adobe" 
FreeSans.ttf and FreeSerif.ttf

I tried every combination of the above list but no change.

tchapi, 2017-12-22:
I know this is an old discussion but I've ran into subpixel alignment problems too using fontconvert and I have made a little handy tool to correct the font if needed — I thought I'd share if it can help others : https://tchapi.github.io/Adafruit-GFX-Font-Customiser/

vonnieda, 2018-11-29:
I've figured out how to revert to the old behavior. In fontconvert.c, after:

```
	// Init FreeType lib, load font
	if((err = FT_Init_FreeType(&library))) {
		fprintf(stderr, "FreeType init error: %d", err);
		return err;
	}
```

Add:

```
	FT_UInt     interpreter_version = TT_INTERPRETER_VERSION_35;
	FT_Property_Set( library, "truetype",
                                  "interpreter-version",
                                  &interpreter_version );	
```

See https://www.freetype.org/freetype2/docs/reference/ft2-properties.html#interpreter-version for documentation.

This makes a huge difference in the clarity of generated fonts, especially small ones.

PaintYourDragon, 2018-12-01:
Merged. Thank you for sorting out the fix for this!

commit 75f81ac3 (2018-12-01): Merge pull request #191 from vonnieda/patch-1

Fix #103 by disabling subpixel rendering.
commit fd68f69d (2018-12-01): Fix #103 by disabling subpixel rendering.

Newer versions of FreeType enable subpixel rendering by default. This makes fonts look terrible on displays that don't support it, like the LCDs this library is targeting. This change disables the subpixel rendering and improves the clarity of fonts.

See https://github.com/adafruit/Adafruit-GFX-Library/issues/103#issuecomment-443031001 for more information.
closed 2018-12-01 by PaintYourDragon