adafruit/Adafruit-GFX-Library #104: Adafruit-GFX still missing drawRGBBitmap

marcmerlin, 2017-01-17:
Howdy,

I have some amount of people using my direct LED Matrix driving code, which supports tricolor LED matrixes, and PWM values for each color.
I've added a method to your library in 2015 to support this, but sadly as of yet, it's still not been included and people are getting the wrong version via the arduino IDE library manager, overriding the version with my patch.
Could you please include it?
https://github.com/adafruit/Adafruit-GFX-Library/pull/39

marcmerlin, 2017-04-06:
@PaintYourDragon I wrote this before it was super useful for your own hardware (although at the time it was already usable on your bi color matrices).
Now that you have Adafruit_NeoMatrix, this code is even more important.
Adafruit_NeoMatrix is capable of 65536 colors in this library, so why restrict users to single color pixmaps?
I've tested this again Adafruit_NeoMatrix and it works perfectly, which is cool since I wrote it 2 years before I had a neomatrix to test with :)
Could I beg you again to include this?  :) Thanks.

mikeysklar, 2026-10-29:
PR #39 was merged, and `drawRGBBitmap()` is in every current release.

It has PROGMEM, RAM and masked versions.

Closing this one.

closed 2026-10-29 by mikeysklar

marcmerlin, 2026-10-29:
@mikeysklar  This was indeed merged in 2017, it somehow didn't ping this issue to remind closing it. 
As a quick update, I have now written a bit over 10 hardware backend to provide Adafruit::GFX and LEDMatrix APIs (+ raw FastLED functions) to those 10-ish hardware backends.
You can find them all here: https://github.com/marcmerlin/Framebuffer_GFX/

I even extended this to work on rPi under linux (so you can run your Adafruit::GFX code on rPi and display in GL, or on an RGBPanel (since rPi can run 3 RGBPanel chains at once as you likely know)
Thank you @PaintYourDragon  and @ladyada  for providing this, this library has spread far and wide 🙂

marcmerlin, 2026-10-29:
Just to show how far the lib has gone, this is my code running on my Framebuffer::GFX lib on 5 Rpis synchronized over Wifi, running Henner's rgbpanels lib to get triple parallel output for better framerate, and support for the new PWM panels.
And all the way at the bottom, is Adafruit::GFX sitting on top of the actual framebuffer (which is RGB888, it does use the conversions from RGB565 to RGB888 before hitting the framebuffer, the same functions you wrote for Adafruit::NeoMatrix back 10-ish years ago).
It may give a smile to @ladyada and @PaintYourDragon , oh and while I'm at it, I'm converting to your new low smaller active-3 RGBPanel board, thank you for that one (even if unfortunately the connectors point up instead of sideways)

Video: https://youtu.be/WkRkH0djN1I

[image]

mikeysklar, 2026-10-29:
Thanks for sharing this, @marcmerlin. 

Those displays look great.