adafruit/Adafruit-GFX-Library #109: Buttons and fonts

HynekBaran, 2017-03-25:
Hi,
I like to set custom fonts on Adafruit_GFX_Button.
Unfortunately, when custom font is set on the fly, buttons does not respect the text position and size. I think that 
```
  _gfx->setCursor(_x1 + (_w/2) - (strlen(_label) * 3 * _textsize),
    _y1 + (_h/2) - (4 * _textsize));
```
shoud be replaced using getTextBounds() code.

HynekBaran, 2017-03-25:
My solution (in Adafruit_GFX_Button::drawButton):
```
  if(_gfx->gfxFont) {
	  int16_t  tx, ty; // HB
	  uint16_t tw, th; // HB
	  _gfx->getTextBounds(_label, _gfx->width()/2, _gfx->height()/2, &tx, &ty, &tw, &th); // HB
	  _gfx->setCursor(_x1 + (_w/2) - tw/2, _y1 + (_h/2) + th/2); // HB
  } else { 		
  	  _gfx->setCursor(_x1 + (_w/2) - strlen(_label)*3*_textsize, _y1 + (_h/2) -  4*_textsize); // HB
  }
```

HynekBaran, 2017-03-25:
But my solution just a hint, consider two buttons with labels "b" and "p" :(

HynekBaran, 2017-03-25:
This is probably slightly better:
```
  if(_gfx->gfxFont) {
      int16_t tx0 =  _gfx->width()/2, ty0=_gfx->height()/2; // HB
	  int16_t  tx, ty; // HB
	  uint16_t tw, th; // HB
	  _gfx->getTextBounds(_label, tx0, ty0, &tx, &ty, &tw, &th); // HB
	  _gfx->setCursor(_x1 + (_w/2) - tw/2 + (tx0-tx), _y1 + (_h/2) - th/2 + (ty0-ty)); // HB
  } else { 		
  	  _gfx->setCursor(_x1 + (_w/2) - strlen(_label)*3*_textsize, _y1 + (_h/2) -  4*_textsize); // HB
  }
```
but the bp problem still remains...

mikeysklar, 2026-10-15:
Resolved with PR #454, which is now merged. Thanks for reporting this!

closed 2026-10-15 by mikeysklar