adafruit/Adafruit-GFX-Library #111: Not even remotely compatable with the ESP32

Fuzzyzilla, 2017-04-02:
In  a project I'm working on, I want to use a adafruit TFT.
In writing the code and compiling for the ESP32 chipset, there were plentiful "May not be initialized" errors, as well as undefined macros (csport, rspinmask, etc)

mattncsu, 2018-04-06:
Working ok for me with a Waveshare 2.9" EPD on ESP32

mikeysklar, 2026-10-21:
The library now builds for ESP32 in CI on every change.
The csport and rspinmask macro errors from this report are gone from the source.
Closing this one.
Please reopen if you still see it on the current release.

closed 2026-10-21 by mikeysklar

Fuzzyzilla, 2026-10-22:
I wrote this issue when I was like 14, I am very sorry for being mean and unhelpful. Thank you all for your hard work!