adafruit/Adafruit-GFX-Library #118: class Adafruit_GFX_Button

mcodo, 2017-04-21:
In class Adafruit_GFX_Button there is a function for detecting if the button is pressed or has been pressed.
But I cant find any documentation for it...
How can I use this?

pacav69, 2017-04-22:
have a look at this sketch
https://learn.adafruit.com/arduin-o-phone-arduino-powered-diy-cellphone/arduin-o-phone-sketch

a snippet from the page
====================
  // create buttons
  for (uint8_t row=0; row<5; row++) {
    for (uint8_t col=0; col<3; col++) {
      buttons[col + row*3].initButton(&tft, BUTTON_X+col*(BUTTON_W+BUTTON_SPACING_X), 
                 BUTTON_Y+row*(BUTTON_H+BUTTON_SPACING_Y),    // x, y, w, h, outline, fill, text
                  BUTTON_W, BUTTON_H, ILI9341_WHITE, buttoncolors[col+row*3], ILI9341_WHITE,
                  buttonlabels[col + row*3], BUTTON_TEXTSIZE); 
      buttons[col + row*3].drawButton();
    }
  }

mcodo, 2017-04-22:
Thanx :-) This is working really good with the provided code.
But when I try on a rotated screen with two columns of four buttons in each column, the touch is not pressing the correct button :-p 
Seams like it's random in a way....

btw: Screen is rotated BEFORE the button setup :-)

Button Setup:

void DrawBtns()
{ 
  for (uint8_t row=0; row<4; row++) {
    for (uint8_t col=0; col<2; col++) {
      buttons[col + row*2].initButton(&tft, BUTTON_X+col*(BUTTON_W+BUTTON_SPACING_X), 
                 BUTTON_Y+row*(BUTTON_H+BUTTON_SPACING_Y), // x, y, w, h, outline, fill, text
                 BUTTON_W, BUTTON_H, ILI9341_WHITE, ILI9341_DARKGREEN, ILI9341_WHITE,
                 buttonlabels[col + row*2], 1); 
      buttons[col + row*2].drawButton();
    }
  }
}

Inside Loop:
void loop() {
  if (ts.touched() ) {
    TS_Point p; 
    if (ts.bufferSize()) {
      p = ts.getPoint(); 
    } else {
      // this is our way of tracking touch 'release'!
      p.x = p.y = p.z = -1;
    }
    
    // Scale from ~0->4000 to tft.width using the calibration #'s
    if (p.z != -1) {
      p.x = map(p.x, TS_MINX, TS_MAXX, 0, tft.width());
      p.y = map(p.y, TS_MINY, TS_MAXY, 0, tft.height());
    }
      
    for (uint8_t b=0; b<8; b++) {
      if (buttons[b].contains(p.x, p.y)) {
        #ifdef DEBUG
          Serial.print("Pressing: "); Serial.println(b);
        #endif
        buttons[b].press(true);  // tell the button it is pressed
      } else {
        buttons[b].press(false);  // tell the button it is NOT pressed
      }
    }

    for (uint8_t b=0; b<8; b++) {
      /*if (buttons[b].justReleased()) {
        #ifdef DEBUG
          Serial.print("Released: "); Serial.println(b);
        #endif
        buttons[b].drawButton();  // draw normal
      }*/
      
      if (buttons[b].justPressed()) {
          buttons[b].drawButton(true);  // draw invert!
          
        if (b == 0) {
            Serial.println("Pressed btn 1"); 
        }
        if (b == 1) {
            Serial.println("Pressed btn 2"); 
        }
        if (b == 2) {
            Serial.println("Pressed btn 3"); 
        }
        buttons[b].drawButton();  // draw normal
        delay(100); // UI debouncing
      }
    }    
  }
}

pacav69, 2017-04-22:
As i understand it
its all to do with arrays:
when in portrait orientation row 0, col 0; row 0 col 1; row 0, col 2 .... etc
then when changed to landscape mode the col's become rows and rows become cols

how the system detects the change in orientation is done by a sensor much like what is found in a mart phone. Which then redraws the screen with the correct button positions.

mikeysklar, 2026-10-21:
Thanks for the report, @mcodo.
Button contains() checks the stored top-left corner and size.
The touch controller does not rotate.
Swap or flip the touch x and y to match your rotation.
This Learn guide explains rotated coordinates: https://learn.adafruit.com/adafruit-gfx-graphics-library/rotating-the-display
Closing this one.

closed 2026-10-21 by mikeysklar