adafruit/Adafruit-GFX-Library #123: problems with small fonts

eadmaster, 2017-05-17:
When using a small font like TomThumb the first line gets printed outside the screen.
(i have to add 13 padding spaces before the string to be printed)

side question:
can i reduce the size of the whitespace between words?
(it is 5 pixels with my 3-pixel width font)

pljakobs, 2017-10-12:
this has been sitting around for a long time.
The dimensions of a font are part of the font definition. So when you say your whitespace is five pixels wide, that's consistent with it's definition 
`{     0,   0,   0,   4,    0,    1 },   // 0x20 ' `
if the following character has an xOffset of 1, that would make it five pixels.
Same goes for the actual height of the font:
they're all defined to be 3 pixels high, but some have an additional offset that would make them bigger.
For fonts that small, I assume that freetype just has issues rendering them onto a small cell without rounding errors. You can certainly go ahead and manually adapt those values, although I figure that would be a whole lot of work without a graphical editor that understands the GFXfont format (of which there's none).

As for why your String is showing outside the screen - you say "first line" but then say you need to pad with spaces - could it be that the first line is printed *above* the visible area? Keep in mind that the flexible fonts are not top left pixel bound but have a baseline that runs at the bottom of the characters with things like p,q, j and y reaching below it. So if you print someting to screen position (0,0) it would be expected that only the the character portions reaching below the baseline would show.

pj

mikeysklar, 2026-11-09:
As @pljakobs said, custom fonts are placed by their baseline, not the top-left corner. With `setCursor(x, 0)` the baseline sits on the top row, so only the descenders show. To put the first line at the top, move the cursor down by the top offset from `getTextBounds()`:

```cpp
int16_t x1, y1;
uint16_t w, h;
tft.getTextBounds(str, 0, 0, &x1, &y1, &w, &h);
tft.setCursor(x, -y1);
```

TomThumb at text size 4 on a Feather HUZZAH ESP8266 and 3.5" HX8357, left `setCursor(10, 0)`, right `setCursor(250, -y1)`:

[image]

<details>
<summary>Sketch</summary>

```cpp
// #123: custom fonts are placed by their baseline, not the top-left corner.
// Left: setCursor(x, 0) puts the baseline on the top row, so only the
// descenders show. Right: move the cursor down by the top offset from
// getTextBounds() and the whole first line is on screen.
// Feather HUZZAH ESP8266 + 3.5" HX8357 FeatherWing.
#include <Adafruit_GFX.h>
#include <Adafruit_HX8357.h>
#include <Fonts/TomThumb.h>

Adafruit_HX8357 tft(0, 15); // CS, DC

const char *str = "Happy day";

void setup() {
  tft.begin(8000000);
  tft.setRotation(3);
  tft.fillScreen(HX8357_BLACK);
  tft.drawFastVLine(240, 0, tft.height(), HX8357_BLUE);

  tft.setFont(&TomThumb);
  tft.setTextSize(4);
  tft.setTextColor(HX8357_WHITE);

  // Left: baseline at y = 0
  tft.setCursor(10, 0);
  tft.print(str);

  // Right: y1 is how far the text reaches above the cursor (negative)
  int16_t x1, y1;
  uint16_t w, h;
  tft.getTextBounds(str, 0, 0, &x1, &y1, &w, &h);
  tft.setCursor(250, -y1);
  tft.print(str);

  tft.setFont();
  tft.setTextSize(2);
  tft.setTextColor(HX8357_YELLOW);
  tft.setCursor(10, 120);
  tft.print("setCursor(10, 0)");
  tft.setCursor(250, 120);
  tft.print("setCursor(250, -y1)");
  tft.setCursor(250, 150);
  tft.print("y1 = ");
  tft.print(y1);
  tft.setCursor(250, 180);
  tft.print("from getTextBounds");
}

void loop() {}
```
</details>

The space width comes from the font data (`xAdvance`), so it can be changed in the font file. Closing, please reopen if this doesn't solve it.

closed 2026-11-09 by mikeysklar