adafruit/Adafruit-GFX-Library #131: Small bitmap and drawBitmap function

iSapozhnik, 2017-08-31:
I have a really strange issue with drawBitmap function. So I'm using Arduino and OLED display SSD1306 i2c 128x32 px. I have a bitmap which is 13x13px. In order to convert it to array I'm using http://javl.github.io/image2cpp/ But unfortunately I can't make it work. I'm getting only some kind of 'snow' instead of my image. The only way how I can make it work is to create 32x32px image and in the top left corner put the icon. But then I have a lot of unnecessary array elements for unused pixels. I have attached the image I'm using (original and 32x32px).
[sources.zip](https://github.com/adafruit/Adafruit-GFX-Library/files/1268304/sources.zip)

Code I'm using:
```
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

#define OLED_RESET LED_BUILTIN  //4
Adafruit_SSD1306 display(OLED_RESET);

#if (SSD1306_LCDHEIGHT != 32)
#error("Height incorrect, please fix Adafruit_SSD1306.h!");
#endif

#define ICON_SIZE  13

const uint8_t fan_off[] PROGMEM = {
  0x07, 0x00, 0x44, 0x02, 0x20, 0x0a, 0x06, 0x73, 0x4d, 0x66, 0x36, 0x33, 0x59, 0x67, 0x30, 0x28, 
0x02, 0x20, 0x11, 0x00, 0x70, 0x7f
  };

void setup() {
  display.begin(SSD1306_SWITCHCAPVCC, 0x3C);
  display.setRotation(2);
  display.clearDisplay();
  display.display();
}

void loop() {
  display.drawBitmap(60, 0, fan_off, ICON_SIZE, ICON_SIZE, 1);
  display.display();
}
```
I would really appreciate for any advice coz I already spent a lot of hours...
Thanks

PaintYourDragon, 2017-09-01:
Increase the image width to the next 8-pixel boundary (e.g. 16x13 pixels), then try image2cpp again. That tool 'packs' the scanline bits (rather than padding to a byte boundary when needed).

closed 2017-09-01 by PaintYourDragon

iSapozhnik, 2017-09-01:
@PaintYourDragon Thanks for you reply. 
So that means that it might be a bug in image2cpp tool. But before Adafruit I was using U8glib library and this image
```
const uint8_t down[] PROGMEM = {
  0x7F, 0x3E, 0x1C, 0x08,
};
```
was perfectly rendered.

PaintYourDragon, 2017-09-01:
Not a bug; different design choices. Some libraries or converters might pad scanlines while others might not. Everything's fine until you cross the streams.

iSapozhnik, 2017-09-01:
Ok, thanks!
Do I need to centre my image or it might be still aligned to left ?

iSapozhnik, 2017-09-01:
@PaintYourDragon thanks for your help! I just tested with a new image sized to reach 8-pixel boundary and it works!