adafruit/Adafruit-GFX-Library #133: print() throws exception on ESP8266 core 2.4.0-rc1

xsrf, 2017-09-05:
### Basic Infos

Hey,
since I upgraded from ESP8266 core 2.3.0 to 2.4.0-rc1 (https://github.com/esp8266/Arduino/releases) I have problems with the Adafruit GFX libraries, which might have to do with the print() implementation of either Adafruit or the esp8266 core - I'm not sure.
The print() function of any library that is based on Adafruit_GFX throws an exception.

### Hardware
Hardware: WeMos D1 Mini / ESP12
ESP8266 Core Version: 2.4.0-rc1
Adafuit GFX Version: 1.2.2

### Description

The print() function of any library that is based on Adafruit_GFX throws an exception.

### Settings in IDE

Module:  			WeMos D1 R2 & Mini
Flash Size: 		4MB/3MB
CPU Frequency:		80Mhz
Upload Using:		Serial/USB

### Sketch

```cpp

#include "Adafruit_GFX.h"
#include "Adafruit_NeoMatrix.h"

Adafruit_NeoMatrix disp = Adafruit_NeoMatrix(128, 64);

void setup()
{
	Serial.begin(115200);
	Serial.println("Boot");
	disp.drawPixel(1, 1, 1);
	Serial.println("drawPixel() OK");
	disp.print("Test"); // throws exception on esp8266 2.4.0-rc1 / works on esp8266 2.3.0
	Serial.println("print() OK");
}

void loop()
{

}

/*
Boot
drawPixel() OK

Exception (3):
epc1=0x40202043 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40233d24 depc=0x00000000

ctx: cont
sp: 3ffef830 end: 3ffefad0 offset: 01a0

>>>stack>>>
3ffef9d0:  feefeffe feefeffe feefeffe feefeffe
3ffef9e0:  00000000 00000000 00000000 00000000
3ffef9f0:  0000ffff feefeffe feefeffe feefeffe
3ffefa00:  00000005 feefeffe feefeffe 000001a4
3ffefa10:  00000008 feefeffe feefeffe 00000001
3ffefa20:  feefeffe feefeffe feefeffe 3ffe883c
3ffefa30:  3ffee8b8 00000000 3ffee8b8 40202320
3ffefa40:  00000001 00000001 3ffe8b89 40203330
3ffefa50:  40201b16 00000000 3ffe8838 40203330
3ffefa60:  3fffdad0 0000000e 3ffefb10 3ffeeaa8
3ffefa70:  3fffdad0 3ffee8b8 3ffee8b8 40202c41
3ffefa80:  3ffe8838 00000001 3ffefb10 40202c6c
3ffefa90:  0001c200 0000001c 3ffefb10 40202c54
3ffefaa0:  3fffdad0 3ffee8b8 3ffefb10 40201ef4
3ffefab0:  feefeffe 00000000 3ffeeaa0 40202960
3ffefac0:  feefeffe feefeffe 3ffeeab0 4010070c
<<<stack<<<

ets Jan  8 2013,rst cause:2, boot mode:(3,6)
*/
```

WereCatf, 2017-09-05:
I do not see you calling disp.begin() Perhaps that is the issue.

xsrf, 2017-09-05:
Adafruit_GFX doesn't have a begin() method... Afaik this only exists on libraries that need some kind of initialization (display controller / IO setup etc..)

WereCatf, 2017-09-05:
You're not using the gfx-library directly. The neomatrix-library does seem 
to have a begin() and at least the examples I saw call that before trying 
to print anything.

xsrf, 2017-09-05:
I know, the above is just an example. 
Actually Adafruit_GFX is derived from [Print](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Print.h#L68) which implements print(). Print is part of the esp core. begin() is implemented in [WString.h](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/WString.h#L211), also part of the esp core... Somehow print(), Serial.print() and disp.print() interfere with eachother... That's too complicated for me :/

xsrf, 2017-09-26:
The issue is confirmed with Arduino core for ESP8266 2.4.0-rc1 and has been corrected after the release of 2.4.0-rc1. Using the latest git version of Arduino core for ESP8266 solves the issue.

Since it's not an adafruit issue, I'll close this one.

closed 2017-09-26 by xsrf