adafruit/Adafruit-GFX-Library #134: Fontconvert for font awesome

emilekm2142, 2017-09-23:
I tried using `fontconvert` to get font awesome icons working on arduino, but it seems like only the ascii character are converted. Can I get the icons too somehow? **Yes, I know that it will not fit on arduino uno**. I have other arduino based board capable of storing that.

pljakobs, 2017-10-12:
I was thinking about icon fonts but I think that will need a slightly different approach than what we have today. Reason being: currently, fontconvert will create a linear structure of pixel matrices for any font along with an array of "character metadata" that is offset the pixel array, width, height etc. This array is contiguous, currently within 7 bit ascii. My wish was always to extend it to utf8 - which per se would be wasteful, but mapping it to ISO character sets would work for the higher spec'd micros. This step would also include non-contiguous mapping tables and probably lend themselves to something like font awesome.
I'd be happy for any help there, I've been rolling around ideas for a while but havn't arrived at a solution that's all of 
- compatible
 (Arduino's printf function does actually provide utf8 characters, so they would have to be mapped back to ISO) 
- performant 
- memory friendly
Also, for icon fonts, I believe there should really be a way to limit the number of icons to what is really needed. You probably don't want every single icon in the set.

So: help would be appreciated, I have ideas but way too many other projects to really push this forward.

emilekm2142, 2017-10-13:
I am not skilled enough to help :/ I hope this feature will work someday.

Bzzz, 2018-03-17:
Well I can convert two separate areas of symbols with a modified version of fontconvert, that's pretty easy. Currently I am baffled where the "wrap around sh!t above 0x7F" code is located in the GFX library, which would at least enable me to re-map characters to the higher areas. I'm doing this to have umlaut characters, but having more character slots than currently available would enable a significant amount of additional FA symbols as well.
Guess it all boils down to fixing #64...

pljakobs, 2018-03-18:
without looking at the code:
There is no wraparound mechanism per se, but the font states the first and last glyph in the set. 
This code then find the correct glyph in the font:
```
uint8_t first = pgm_read_byte(&gfxFont->first);
if((c >= first) && (c <= (uint8_t)pgm_read_byte(&gfxFont->last))) {
    GFXglyph *glyph = &(((GFXglyph *)pgm_read_pointer(
      &gfxFont->glyph))[c - first]);
```
The code then checks for width & height to be greater than zero which indicates that the glyph has an associated bitmap (this is a hack allowing to have essentially empty glyphs without having to store empty bitmaps) and if so, will draw the bitmap.

I would assume that your modified fontconvert might not set _first_ and _last_ correctly?

pj

Bzzz, 2018-03-18:
That one?

```
const GFXfont FreeMono18pt8b PROGMEM = {
  (uint8_t  *)FreeMono18pt8bBitmaps,
  (GFXglyph *)FreeMono18pt8bGlyphs,
  0x20, 0xFC, 35 };
```

(this is the quick fix from #64)
The modified fontconvert does set first and last kind-of correct, depending on your way of counting. It starts at 0x20 if not modified, and ends at the designated last char that is included. So start and end do match the original range, but obviously there aren't enough bitmaps to fill that entire range - otherwise even small fonts can blow up to several megabytes. That however does not explain why stuff wraps around - how does the script know there is a gap when there is no gap in the bitmap table itself?

I've moved on for now and I've been using said trick from KeeTraxx to get a prototype display "working" with a lot of separate lines inserting umlauts. I actually might need some icons from FA or other icon fonts for the next steps, and chopping text to insert higher ASCII chars is pretty annoying, so I'm still interested in getting that issue resolved.

pljakobs, 2018-03-18:
@Bzzz there are two structures: the Glyph structure (GFXglyph) which has to be a contiguous array of GFXglyph structures 

```
typedef struct { // Data stored PER GLYPH
	uint16_t bitmapOffset;     // Pointer into GFXfont->bitmap
	uint8_t  width, height;    // Bitmap dimensions in pixels
	uint8_t  xAdvance;         // Distance to advance cursor (x axis)
	int8_t   xOffset, yOffset; // Dist from cursor pos to UL corner
} GFXglyph;
```
and the Font structure containing the bitmaps and the global information

```
typedef struct { // Data stored for FONT AS A WHOLE:
	uint8_t  *bitmap;      // Glyph bitmaps, concatenated
	GFXglyph *glyph;       // Glyph array
	uint8_t   first, last; // ASCII extents
	uint8_t   yAdvance;    // Newline distance (y axis)
} GFXfont;
```
The write() function, at it's core, is this:
```
if(c == '\n') {
            cursor_x  = 0;
            cursor_y += (int16_t)textsize *
                        (uint8_t)pgm_read_byte(&gfxFont->yAdvance);
        } else if(c != '\r') {
            uint8_t first = pgm_read_byte(&gfxFont->first);
            if((c >= first) && (c <= (uint8_t)pgm_read_byte(&gfxFont->last))) {
                GFXglyph *glyph = &(((GFXglyph *)pgm_read_pointer(
                  &gfxFont->glyph))[c - first]);
                uint8_t   w     = pgm_read_byte(&glyph->width),
                          h     = pgm_read_byte(&glyph->height);
                if((w > 0) && (h > 0)) { // Is there an associated bitmap?
                    int16_t xo = (int8_t)pgm_read_byte(&glyph->xOffset); // sic
                    if(wrap && ((cursor_x + textsize * (xo + w)) > _width)) {
                        cursor_x  = 0;
                        cursor_y += (int16_t)textsize *
                          (uint8_t)pgm_read_byte(&gfxFont->yAdvance);
                    }
                    drawChar(cursor_x, cursor_y, c, textcolor, textbgcolor, textsize);
                }
                cursor_x += (uint8_t)pgm_read_byte(&glyph->xAdvance) * (int16_t)textsize;
            }
        }
```
As you can see, there is no wraparound, at least not intentionally. I am wondering if there might be some overflow with the uint8_t type of the variable 'c' passed to the function.
If c is < than 'first' or larger than 'last', nothing at all should happen, thought.
the first element in the GFXGlyph array must be 8Bit code 'first' because it's being read as element [c -
 first].

I cannot find any reason why anything would wrap around if 0<=c<=255

mikeysklar, 2026-11-12:
Font Awesome icons are at U+F000 and up, so pass that range to fontconvert. GFX reads one byte per character, so then move the font down to 8-bit codes:

1. `fontconvert fa-solid-900.ttf 24 61440 61480 > FASolid24pt.h` (U+F000 to U+F028, about 10 KB)
2. In the header, change the last line's `0xF000, 0xF028` to `0x80, 0xA8`.
3. Byte `0x80 + n` is now icon U+F000 + n:

```cpp
display.setFont(&FASolid24pt);
display.print("\x95"); // U+F015 house
```

One font can hold up to 128 icons this way (0x80 to 0xFF). Font Awesome Free 6 solid, Feather HUZZAH ESP8266 + 3.5" HX8357 FeatherWing:

[image]

Closing, please reopen if you have trouble with other icon ranges.

closed 2026-11-12 by mikeysklar