adafruit/Adafruit-GFX-Library #137: Abstract consumer

spmp, 2017-09-29:
Is it possible to provide an abstract consumer of this class such that a library can depend on a device that draws functions using the `Adafruit-GFX-Library` primatives without the need for a concrete implementation. 
For example, I have a project that currently is using an `Adafruit_SSD1306` device, and has some class that takes this device in its constructor in order to draw things.
 I anticipate that this device will become obsolete (for me) soon, and as such I would like my class to take a generic display device, independent of device or bus.
 I cannot see how to do this currently and am assuming that this class is the one to provide a generic/abstract type giving me this functionality, otherwise does there need to be a new class/library which is this generic/abstract type?

Cheers.

pljakobs, 2017-09-29:
I am not 100% sure if this answers your question, but I'll give it a shot:
Currently, I'm writing a gauges class that abstracts the display of values on any kind of (currently) Adafruit_GFX display. The code is [here](https://github.com/pljakobs/Arduino_Gauges).
In it's simplest form, this is instantiated using
```
eisplayGauge::displayGauge(Adafruit_GFX *display){
	_x=_y=_w=_h=0;
	_display=display;
	_visible=true;
}
``` 
and it then uses the internal pointer
`    Adafruit_GFX  *_display;`
to access the actuall display hardware. This pointer is initiated with the pointer to a proper display.

If the gauges library is useful for you, this could be the way to go (although it should be said that for that, you really want a 64kB RAM device like a Teensy3), otherwise, I guess you can probably implement something in a similar manner.

pj

spmp, 2017-10-01:
Thank you! I will look more into your solution
 And my bad, I see now that the Adafruit displays are _all_ `Adafruit_GFX` - which is what I was after in the first place 8)

closed 2017-10-01 by spmp

neu-rah, 2017-10-01:
yes it is possible, I'm using Adafruit_GFX as an abstract GFX class to fit all specific implementations on my menu library, also used on gauges and many other stuff.

my adafruit gfx driver hook: https://github.com/neu-rah/ArduinoMenu/blob/master/src/menuIO/adafruitGfxOut.h#L25

there are some examples of using that with multiple specific implementations.

spmp, 2017-10-01:
> ArduinoMenu/blob/master/src/menuIO/adafruitGfxOut.h#L25

Too good!
Thanks for the examples guys, both of which look like I will want to use 8)
 Its a pity we cannot see all the projects with `Adafruit-GFX-Library` as
dependency - well, more projects that extend it for display creation 8)