adafruit/Adafruit-GFX-Library #138: Fonts

VyacheslavKH, 2017-10-11:
Good afternoon. When loading new fonts it is not possible to change the color of the substrate on which they are output. If the text is displayed in the same place on the screen, then the old characters are not overwritten.
  tft.setTextColor (Blue, Black); The color of the text changes, but there is no substrate. Please tell me how you can solve this problem.

pljakobs, 2017-10-11:
This is expected behaviour as proportional fonts don't have defined bounding boxes and characters may actually overlap, so clearing the character box might iteself lead to unintended behaviour.
You can use getTextBounds to get the actual bounding box for the given string (or the old string or the maximum string) and use fillRect to clear the text area. 
Alternatively, you can create an extra canvas in the correct size, render your text on that canvas and then copy the canvas to your display device. 
I have used the latter in my Gauges library (https://github.com/pljakobs/Arduino_Gauges) if you're looking for an example. The advantage is, that the whole text field will be displayed in one go.

mikeysklar, 2026-11-09:
This is by design: custom fonts don't draw a background, because glyphs can overlap. The README lists the two ways to do it, under the "will NOT be merged" list: clear the area with `fillRect()`, or (no flicker) draw the text into a `GFXcanvas1` and copy it with `drawBitmap()` and a background color.

Here is a counter in FreeSans18pt updating every 250 ms on a Feather HUZZAH ESP8266 and 3.5" HX8357. Left is `setTextColor(fg, bg)` only, right is the canvas method:

[image]

<details>
<summary>Sketch</summary>

```cpp
// #138: updating custom-font text in place.
// Left: setTextColor(fg, bg) only. Custom fonts draw no background, so old
// digits stay on screen. Right: draw into a GFXcanvas1, then copy it with
// drawBitmap() and a background color, which erases the old text without
// flicker (README, "pull requests ... will NOT be merged").
// Feather HUZZAH ESP8266 + 3.5" HX8357 FeatherWing.
#include <Adafruit_GFX.h>
#include <Adafruit_HX8357.h>
#include <Fonts/FreeSans18pt7b.h>

Adafruit_HX8357 tft(0, 15); // CS, DC
GFXcanvas1 canvas(200, 50);

void setup() {
  tft.begin(8000000);
  tft.setRotation(3);
  tft.fillScreen(HX8357_BLACK);
  tft.setTextSize(2);
  tft.setTextColor(HX8357_YELLOW);
  tft.setCursor(20, 20);
  tft.print("setTextColor");
  tft.setCursor(20, 40);
  tft.print("(fg, bg)");
  tft.setCursor(260, 20);
  tft.print("GFXcanvas1 +");
  tft.setCursor(260, 40);
  tft.print("drawBitmap()");
  canvas.setFont(&FreeSans18pt7b);
}

void loop() {
  static uint16_t n = 0;
  n = (n * 7 + 13) % 1000; // jumpy values so every digit changes

  // Left: font has no background, so this just piles digits on top
  tft.setFont(&FreeSans18pt7b);
  tft.setTextColor(HX8357_WHITE, HX8357_BLACK);
  tft.setTextSize(1);
  tft.setCursor(20, 160);
  tft.print(n);

  // Right: canvas is cleared, then copied with fg and bg colors
  canvas.fillScreen(0);
  canvas.setCursor(0, 38);
  canvas.print(n);
  tft.drawBitmap(260, 122, canvas.getBuffer(), canvas.width(),
                 canvas.height(), HX8357_WHITE, HX8357_BLACK);

  tft.setFont();
  delay(250);
}
```
</details>

Closing since the README covers it, please reopen if the canvas method doesn't work for you.

closed 2026-11-09 by mikeysklar