adafruit/Adafruit-GFX-Library #143: How to get perfect text width of a scrolling text ?

Jarvis785, 2017-12-16:
Suppose in a scrolling text i want to scroll a string "Hiiiiiiiiiiiiiiiiiii"  . So, to find scrolling text displacement i have to do 20(total number of char)*6(width of a single char) =120

But actual width of char 'i' is 3 but i assumed it 6 so output 120 is not a accurate value

Is there any function is there or any user defined function to find actual width of each character of a font and also width of a string.

Please help me out guy :(

pljakobs, 2017-12-16:
I guess the most compatible way would be to call getTextBounds() to get the absolute width of your string and use that value.

Jarvis785, 2017-12-16:
How to call getTextBounds() .
Can you please give an example .

pljakobs, 2017-12-16:
[code]
char*  buf[]='Hiiiiiiiiiiiiiiiiiii';
int16_t x1,y1,x2,y2,width;
display->getTextBounds(_buf, 0, 0, &x1, &y1, &x2, &y2);
width=x2-x1;
[/code]

at least I think this should work.

Jarvis785, 2017-12-16:
Thank you so much :)
A little tweak and it worked for me !!

This code works
char buf[]="Hiiiiii";
int16_t x1,y1,x2,y2,width;
matrix.getTextBounds(buf, 0, 0, &x1, &y1, &x2, &y2);
width=x2-x1;

pljakobs, 2017-12-17:
you are of course right. My fault right there! 

pj

Jarvis785, 2017-12-17:
No Problem :)  from you hint only i tweaked a little .. ;)

closed 2017-12-17 by Jarvis785