adafruit/Adafruit-GFX-Library #144: Degree symbol with TomThumb font

inrepublica, 2017-12-24:
I'm trying to use TomThumb with my little project. I use a wemos d1 mini pro (ESP8266) and the wemos OLED shield (64*48 SSD1306). Here is my project -> [my project](https://github.com/inrepublica/Wemos_test_oled_texte)

When i use the default font no problem degree symbol is used:
` display.setFont();
  display.setTextSize(1);
  display.setTextColor(WHITE);
  display.setCursor(0,10);
  display.print(t);display.print("\xA7");display.print(" ");display.print(h);display.print("%");`

But when i use the tomthumb font, nothing is displayed:
` display.setFont(&TomThumb);
  display.setTextSize(1);
  display.setTextColor(WHITE);
  display.setCursor(0,47);
  display.print(t);display.print("\xB0");display.print(" ");display.print(h);display.print("%");
  display.display();`

[image]

Sorry for my bad english ;-(

baryluk, 2018-06-04:
FYI. There is TomThumb in the official repos, so try including that. And before including, make sure to do

```
#define TOMTHUMB_USE_EXTENDED 1
```

it will use a bit more space, but give you some more characters, including degree symbol.

I still do not know why there is a vertical line below degree symbol tho.

Spartrap, 2024-09-13:
This is an old thread but as the situation is still the same, I'll add some remarks:

- TomThumb doesn't contain all extended characters, there's a gap in the middle, so you have to adjust your e.g. ISO 8859-x encoded string.

Here is a snippet I use to fix it:

    // the TomThumb font is shipped with Adafruit GFX, and is the only official font that has extended characters
    // but it's somewhat quirky: it misses chars from 0x7E to 0xA1.
    // It also has a dozen extra chars above 0xff that I didn't bother to add except for the Euro symbol
    if (applyTomThumbShift) {
      out = (unsigned char *)output;
      while (*out) {
        if (*out > 0xA0U)
          if (*out == 0xA4U)
             *out = 0xEAU; // add Euro char €
           else
             *out -= 0x22U; // shift the codepoint to make up for the gap
        else if (*out > 0x7EU)
          *out = 0x20U; // missing characters. Convert to space.
        out++;
      }
    }

Also, as mentioned, on top of the TOMTHUMB_USE_EXTENDED flag, you have to modify the last line of TomThumb.h :

const GFXfont TomThumb PROGMEM = {(uint8_t *)TomThumbBitmaps,
                                  (GFXglyph *)TomThumbGlyphs, 0x20, 0xEA, 6};

HIH.

closed 2026-11-12 by mikeysklar