adafruit/Adafruit-GFX-Library #145: new SPITFT code breaks React Counter example

dhalbert, 2018-01-08:
A user was unable to compile this code: https://github.com/adafruit/TrinketReactCounter/tree/master/TrinketReactCounter_7segment with the latest version of the GFX library. I verified. See https://forums.adafruit.com/viewtopic.php?f=52&t=129056. Works fine with the 1.2.2 release available from the Library Manager, but not with the latest commit on master. (The user had originally cloned the library using git.)

PaintYourDragon, 2018-01-08:
Fixed, new release pushed to Github. Some code had been added along the way that was not ATtiny-compatible. Fix just disables this code on ATtiny; color graphics is a bit out of their scope anyway.

closed 2018-01-08 by PaintYourDragon

ladyada, 2018-01-08:
woo you got to it before me :D thanks!