adafruit/Adafruit-GFX-Library #147: rendering misshap in fillCircleHelper

noxobscura, 2018-01-20:
System: 2.8" TFT Touch Shield for Arduino w/Capacitive Touch + Arduino Uno R3 (smd)

Description: I noticed a strange behaviour (i.e. one line is drawn horizontally straight to the right border of the display) while using the Adafruit_GFX_Button class. I was abled to track it down to fillCircleHelper (see attached file). Changing the x or y pos only by one pixel avoids this behavior.

Can someone verify this behavior or is it just my system?
. 
[2017-12-18_bug02.ino.txt](https://github.com/adafruit/Adafruit-GFX-Library/files/1648501/2017-12-18_bug02.ino.txt)

noxobscura, 2018-01-20:
As the files seemed to be modified during upload I added the zipped version here: 
[2017-12-18_bug02.zip](https://github.com/adafruit/Adafruit-GFX-Library/files/1648502/2017-12-18_bug02.zip)

pljakobs, 2018-08-01:
I have tried to reproduce the issue, albeit on a very different platform (a Teensy) and found no drawing errors. It does sound pretty similar to [issue 165](https://github.com/adafruit/Adafruit-GFX-Library/issues/164) though, so I wonder how robust the GFX class is when it comes to rounding errors of the underlying platform.

mikeysklar, 2026-10-29:
Tested your sketch's exact call on current master with a 320x240 canvas.
Every line it draws stays inside the button.

| test | lines outside rect |
|---|---|
| fillCircleHelper(287,162,12,1,25) | 0 |
| same with the Jan 2018 code | 0 |
| 2.6M round rects and buttons | 0 |

16-bit int (Uno) math gives the same values.
Closing this one.
Please reopen with the exact coordinates if you still see it.

closed 2026-10-29 by mikeysklar