adafruit/Adafruit-GFX-Library #148: Fonts compression explanation

ihouses, 2018-01-21:
Hello,

I would like to understand how the fonts are compressed. Reading the code I see that there is a pointer in GFXglyph that points to the beginning of the  compressed character within the GFXfont structe. What I don't understand is how the bytes are interpreted. Is there any explanation or tutorial regarding this topic anywhere?

Thanks by advance.

pljakobs, 2018-01-23:
your best bet would be to look at the fontconvert code, basically, the format uses the minimum amount of bytes to represent all pixels that make up a character.

pj

ihouses, 2018-01-24:
Hi,

I have check the code but I really dont't undertand it. Since  I have been reading regarding the font compression and the idea is when the byte that you read from the buffer has the most significant bit to 1 it means that the rest of the bits (the 7 remaning bits) are the pixel information, otherwise if the bit is 0 it means that there a long list of repeated pixels (same value).  Am I right with this?

But when I check the code I see that sometimes is calling the function draw_rectangle with a square shape  of (size x size).  :?

pljakobs, 2018-01-24:
It's been a while since I wrote this code and I think there was some enhancement to it, too.
However, I just looked at both the fontconvert as well as the drawChar code and I believe you're confusing two things. The bit fields for the gfxFont are just that: plain bits, no run length encoding (I've done that for some bitmap graphics, but not here, since usually, fonts are not that highly repetitive. Encoding the bits would mean that every byte can only hold seven bits, so from the get-go, the format would be 12,5% less efficient. For fonts less than 16px wide, I guess the compression would be negative)
What you see is the crude scaling mechanism that adafruitGFX implements where you can set a 'size' parameter which results on every pixel being scaled by a factor of 'size' - hence the drawRect() s

PaintYourDragon, 2018-01-24:
There is no compression in the font data. Each glyph is an uncompressed bitmap, fitted to the individual glyph bounds, rows padded to a byte boundary where necessary (width & 7 != 0).

mohammadkhoshsiar, 2018-01-26:
hi i am persian and use fontconvert.exe for persian font. but adress off alphabet in windows character map are diffrent with glyph in font.h   how i can use persian font in with adres off charactermap
for example  in character map ع adress is U+ 0639  but in .h file is not

pljakobs, 2018-01-26:
Hi Mohammad, in short: currently, the implementation in AdafruitGFX by default only handles a seven bit charset. You can create larger font files to contain more glyphs, but that will only work über to character position 255. 
If you want to continue this discussion, please open a new issue for it, this one was concerning the actual storage format, you are asking for utf-8 and international character support.

PaintYourDragon, 2018-01-26:
utf-8 support will probably never happen in GFX. This all has to fit on an AVR.

Plan was to support extended character sets with a 'code page' approach (i.e. multiple 8-bit fonts as needed). While it's inelegant and harkens back to DOS-type stuff, it's within the scope of the AVR, the approach is still used in memory-constrained devices such as thermal receipt printers.

GFX could handle this right now, I think, but none of the required code hasn't been implemented in fontconvert. This would need to pluck individual glyphs out of a larger font and assign them slots between 1 and 255 in a GFX raster font. Even the means of specifying the inputs and outputs has not been defined.

pljakobs, 2018-01-26:
@PaintYourDragon well... what I have (somewhere) is some very basic code to map iso codepages to utf-8 and vice versa. I believe while we could theoretically do something simpler (like: having an interface that only deals with single byte characters), the way the print functions work, we do get utf-8, thus mapping to/from iso 8859 character sets would preserve the most "natural" experience for anyone writing code.
One of the things that held me back is that for the more complex utf->iso mappings, we'll need quite a few if statements, potentially slowing down the code and making it difficult to read.
It's still on my list though.
Where are those pixies that write your code, when you ever need them?
pj

ihouses, 2018-01-27:
Thanks a lot for you comments. I was very confused because I have read somewhere that the fonts where compressed. But now everything works fine. 

Thanks again.

eltomjan, 2018-11-28:
I made compression a bit better than LZW, but pull request is waiting more than year yet...
You can check it in my VS C++ .Net/ Windows simulator:
https://github.com/eltomjan/ETEhomeTools/tree/master/ArduinoSimulator
Or directly in forked library https://github.com/eltomjan/Adafruit-GFX-Library
In case you like 40-50% smaller fonts...
There are 2 quite simple tricks now - I am using XOR between rows (bottom2top when creating font and top2bottom during print) to increase #of zeros and then there is a split to block logic now (whole char bitmap is taken as blocks of particular bit size and in case block has only zero bits, it is replaced by 0 bit, if there are any non-zeros, then data are encoded by 1 high bit and original data), each char starts with 1B holding block size. Rest works same as common GFX fonts - bits are decoded during print and sent as before (columns, rows from top-left corner). There is only row buffer added to be able to (Un-)XOR back data of current line.

eltomjan, 2018-12-08:
Finished also encoding and there is also very simple reader and "editor" in JScript if someone want to play with GFX format without need of toolset except browser and ev. debugger. Now have also very simple GFX editor (show/update/save char, recreate font) in JScript (legacy format + convertion of glyphs table option) here https://github.com/eltomjan/ETEhomeTools/blob/master/HTM_HTA/GFX/fontShowConvert.htm

antonysigma, 2025-03-03:
A trick that worked for me is to filter the font map at compile time. It is not strictly "compression", but typically I can achieve around to `128 characters / 15 characters = 850 times` compression ratio which this technique. 

https://github.com/antonysigma/Adafruit-GFX-Library/commit/3cf9a78494ab3e0018eaf6ebe35744fc7bc1d748

Let's say the GFX library is used for a dashboard application to monitor room temperature and humidity. The firmware and the compiler should already know the list of used characters (i.e. `+-0123456789.%C`). So we can to extract just those characters at compile-time with C++17 style code. And with quite a few `constexpr std::array` and `constexpr switch statement` black-magic.

Firmware utilization report from [Google Bloaty](https://github.com/google/bloaty):
```
    FILE SIZE        VM SIZE    
 --------------  -------------- 
  48.2%  18.8Ki  22.8%  5.62Ki    [127 Others]
   5.9%  2.31Ki   9.2%  2.26Ki    Adafruit_SSD1306::display()
   5.6%  2.19Ki   8.7%  2.14Ki    Adafruit_GFX::drawChar()
   5.0%  1.93Ki   7.6%  1.88Ki    TwoWire::endTransmission()
   3.3%  1.28Ki   5.0%  1.22Ki    Adafruit_GFX::drawFastVLine()
   3.1%  1.22Ki   4.7%  1.17Ki    Adafruit_GFX::writeLine()
   3.0%  1.15Ki   4.4%  1.09Ki    Adafruit_SSD1306::begin()
   2.8%  1.08Ki   4.1%  1.02Ki    Adafruit_SSD1306::ssd1306_commandList()
   2.3%     922   3.4%     868    Adafruit_GFX::drawFastHLine()
   2.2%     876   3.4%     848    __vector_24
   2.1%     833   3.1%     778    Adafruit_SSD1306::invertDisplay()
   2.2%     885   2.8%     719    splash1_data
   1.9%     754   2.8%     696    Adafruit_SSD1306::drawFastHLine()
   1.9%     750   2.7%     692    Adafruit_SSD1306::ssd1306_command1()
   1.8%     711   2.7%     690    main
   1.9%     746   2.7%     688    Adafruit_SSD1306::drawFastVLine()
   1.8%     721   2.5%     630    Adafruit_GFX::drawBitmap()
   1.3%     537   2.0%     512    twi_stop
   1.3%     514   1.9%     480    splash2_data
   1.2%     480   1.7%     438    Adafruit_GFX::write()
   1.2%     480   1.7%     430    Adafruit_GFX::writePixel()
...
   0.4%     139   0.3%     70    (anonymous namespace)::FilteredFontMap<>::font_data
 100.0%  38.9Ki 100.0%  24.7Ki    TOTAL
```

Note that the filtered font data takes only `(15 - 1) characters * 5 byte/char = 70 bytes` to encode `+-0123456789.%C`. The compiler even inlined the default character `C` into the main program, saving extra 5 bytes. It is far smaller than the Adafruit's default splash screen `719 + 480 bytes`.

mikeysklar, 2026-11-14:
The glyph bitmaps are packed, not compressed.

Each glyph is 1 bit per pixel, row by row, most significant bit first.

Rows are not padded, so a row can end partway through a byte.

Each glyph starts on a new byte at `bitmapOffset`.

| `GFXglyph` field | Meaning |
|---|---|
| `bitmapOffset` | First byte of this glyph in `bitmap[]` |
| `width`, `height` | Size of the bitmap in pixels |
| `xAdvance` | How far the cursor moves after this glyph |
| `xOffset`, `yOffset` | Top left of the bitmap, relative to the cursor on the baseline |

Decoding one glyph by hand:

```cpp
GFXglyph *g = &font->glyph[c - font->first];
const uint8_t *bmp = font->bitmap + g->bitmapOffset;
uint8_t bits = 0;
for (int i = 0, y = 0; y < g->height; y++)
  for (int x = 0; x < g->width; x++, i++) {
    if ((i & 7) == 0) bits = pgm_read_byte(&bmp[i / 8]);
    if (bits & 0x80) drawPixel(cx + g->xOffset + x, cy + g->yOffset + y);
    bits <<= 1;
  }
```

`R` from FreeSans12pt7b is 14 pixels wide, so its rows cross byte boundaries.

Library on the left, the loop above on the right, Feather HUZZAH ESP8266 + 3.5" HX8357 FeatherWing:

[image]

Closing, please reopen if anything is unclear.

<details>
<summary>Test sketch</summary>

```cpp
// #148: decode one GFX font glyph by hand and compare with the library.
// Feather HUZZAH ESP8266 + 3.5" HX8357 FeatherWing.
#include <Adafruit_GFX.h>
#include <Adafruit_HX8357.h>
#include <Fonts/FreeSans12pt7b.h>

Adafruit_HX8357 tft(0, 15); // CS, DC
const int S = 8;            // magnification

void setup() {
  Serial.begin(115200);
  tft.begin();
  tft.setRotation(3);
  tft.fillScreen(HX8357_BLACK);

  const GFXfont *f = &FreeSans12pt7b;
  char c = 'R'; // 14 wide, so rows cross byte boundaries
  GFXglyph *g = &f->glyph[c - f->first];
  const uint8_t *bmp = f->bitmap + g->bitmapOffset;

  // Library: baseline at y = 250, glyph box starts at xOffset, yOffset.
  tft.setFont(f);
  tft.setTextSize(S);
  tft.setTextColor(HX8357_WHITE);
  tft.setCursor(40 - g->xOffset * S, 250);
  tft.print(c);

  // By hand: 1 bit per pixel, MSB first, rows not padded.
  int x0 = 280, y0 = 250 + g->yOffset * S;
  uint8_t bits = 0;
  for (int i = 0, y = 0; y < g->height; y++) {
    for (int x = 0; x < g->width; x++, i++) {
      if ((i & 7) == 0)
        bits = pgm_read_byte(&bmp[i / 8]);
      bool on = bits & 0x80;
      bits <<= 1;
      tft.drawRect(x0 + x * S, y0 + y * S, S, S, 0x2104);
      if (on)
        tft.fillRect(x0 + x * S + 1, y0 + y * S + 1, S - 1, S - 1,
                     HX8357_YELLOW);
      Serial.print(on ? '#' : '.');
    }
    Serial.println();
  }

  tft.setFont();
  tft.setTextSize(2);
  tft.setTextColor(HX8357_WHITE);
  tft.setCursor(40, 20);
  tft.print("tft.print('R')");
  tft.setCursor(280, 20);
  tft.print("decoded by hand");
  tft.setTextColor(HX8357_YELLOW);
  tft.setCursor(280, 270);
  int nbits = g->width * g->height;
  tft.printf("%dx%d = %d bits", g->width, g->height, nbits);
  tft.setCursor(280, 292);
  tft.printf("in %d bytes", (nbits + 7) / 8);
  Serial.printf("#148 glyph %dx%d offset %d\n", g->width, g->height,
                g->bitmapOffset);
}

void loop() {}
```

</details>

closed 2026-11-14 by mikeysklar

eltomjan, 2026-11-14:
Added my compression to existing GFX editor:
https://eltomjan.github.io/JStoolsSPAdemos/Adafruit-GFX-Font-Customiser.htm

Simulated demo project:
https://wokwi.com/projects/394882822504014849