adafruit/Adafruit-GFX-Library #149: Why does getTextBounds() not take a const char*?

Sod-Almighty, 2018-02-07:
I cannot pass a `const char*` to `getTextBounds()`. Why not?

pljakobs, 2018-02-07:
'''
getTextBounds(char *string, int16_t x, int16_t y,
      int16_t *x1, int16_t *y1, uint16_t *w, uint16_t *h),
    getTextBounds(const __FlashStringHelper *s, int16_t x, int16_t y,
      int16_t *x1, int16_t *y1, uint16_t *w, uint16_t *h);
'''
It's only defined for char* and flash strings. If you define a const char*, it's probably best to define it as flash string anyway, that way it'll reside in progmem which is larger on most platforms. 

pj

Sod-Almighty, 2018-02-08:
Perhaps I should be more specific. I already _knew_ the function signatures; what I _meant_ was "why didn't you make the function accept a `const char*` instead, given that it doesn't modify the string?"

I'm using an ESP8266. Not only does it have plenty of RAM, it also doesn't seem to properly support the `Flash` library or the `F()` macro.

Sod-Almighty, 2018-02-10:
I have never heard of such a limitation. But in any event, nothing stops there being two versions of the function, one const and one non-const. Having to explicitly cast away constness is _bad programming practice_.

pljakobs, 2018-02-10:
@Sod-Almighty I'm sure you are aware of the fact that this is open source software being developed in a collaborative way. You are of course free to for it, write the suitable overloads for your needs, test them and create a PR.

vortigont, 2018-04-22:
I've done this PR #130 quite some time ago, it also allows to pass String() to getTextBounds()

ladyada, 2018-08-12:
https://github.com/adafruit/Adafruit-GFX-Library/pull/130#issuecomment-412334177

mikeysklar, 2026-10-21:
Thanks for the report, @Sod-Almighty.
getTextBounds takes a const char* since release 1.2.8, from PR #130.

It also takes a String and a flash string.

Closing this one.

closed 2026-10-21 by mikeysklar