adafruit/Adafruit-GFX-Library #153: Cannot set different font after empty setFont

cmmonaco, 2018-03-07:
When attempting to changing font after calling setFont(), the font remains the default gfxfont and will not change again no matter how many times setFont(&newfont) is called. (newfont is just a placeholder for any of the included fonts).

PaintYourDragon, 2018-03-07:
Can't reproduce; switching back and forth as expected.

What architecture? And can you provide a minimal example demonstrating the issue?

cmmonaco, 2018-03-07:
I'm using an Arduino Uno R3 and the 3.5" TFT touchscreen.

`#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_TFTLCD.h> // Hardware-specific library
#include <TouchScreen.h>
#include <Fonts/FreeSans18pt7b.h>
#include <Fonts/FreeSans12pt7b.h>
#include <Fonts/FreeSans9pt7b.h>

//LCD Control Pins
#define LCD_CS 13 // Chip Select goes to Analog 3
#define LCD_CD 12 // Command/Data goes to Analog 2
#define LCD_WR 11 // LCD Write goes to Analog 1
#define LCD_RD 10 // LCD Read goes to Analog 0 

#define Y_MINUS A1
#define X_MINUS A0
#define Y_PLUS A3
#define X_PLUS A2

#define MINPRESSURE 10
#define MAXPRESSURE 1000
#define TS_MINX 950
#define TS_MAXX 91
#define TS_MINY 127
#define TS_MAXY 900

#define  BLACK   0x0000
#define BLUE    0x001F
#define RED     0xF800
#define GREEN   0x07E0
#define CYAN    0x07FF
#define MAGENTA 0xF81F
#define YELLOW  0xFFE0
#define WHITE   0xFFFF

#define SENSOR_PIN A5

//Define states and modes
enum state {
  IDLE,
  RUNNING_TIMER,
  RUNNING_HOLD,
  SET_TIME,
  SET_PRESSURE
};
volatile state currentState = IDLE;
state previousState = -1;

enum mode {
  TIMER,
  HOLD
};
mode currentMode = HOLD;

byte isPressed = 0;
long int stateChangeTime = 0;

//Function prototypes
void idle_state();
void running_hold_state();
void running_timer_state();
void set_time_state();
void set_pressure_state();

volatile int pressed = 0;
volatile float pressure = 0.0;
int setPoint = 0;
long setTime = 0;
volatile long currentTime = 0;

//define touchable areas on screen as a rectantnle with diagonal points
//defined by {x1,y1,x2,y2}
const int PRESSURE_SET_AREA[] = {0, 0, 340, 200};
const int TIME_SET_AREA[] = {90, 200, 360, 320};
const int START_STOP_AREA[] = {361, 200, 480, 320};
const int MODE_CHG_AREA[] = {0, 200, 90, 320};

Adafruit_TFTLCD tft(LCD_CS, LCD_CD, LCD_WR, LCD_RD, A4);
TouchScreen ts = TouchScreen(X_PLUS, Y_PLUS, X_MINUS, Y_MINUS, 278);

void setup()
{
  //Serial.begin(9600);
  
  //Get Screen Ready
  tft.reset();
  uint16_t identifier = tft.readID();
  tft.begin(identifier);
  tft.setRotation(1); //set to landscape

  //Draw start-up message
  //tft.fillScreen(WHITE);
  //Draw font "Incubator Controller v1.0"

  //Draw all static elements
  //Divide up Screen
  tft.fillScreen(BLACK);
  tft.drawFastHLine(0, 200, 480, WHITE);
  tft.drawFastVLine(360, 200, 120, WHITE);
  tft.drawFastVLine(90, 200, 120, WHITE);
  tft.drawFastVLine(340, 0, 200, WHITE);

  //Draw static text
  tft.setFont(&FreeSans18pt7b);
  tft.setCursor(265,135);
  tft.print("KPa");
  tft.setCursor(385, 80);
  tft.print("ON");
  tft.setCursor(375, 160);
  tft.print("OFF");
  
  tft.setFont();
  tft.setCursor(5,30);
  tft.setTextSize(15);
  tft.print("000"); //tft.print(pressure);
  
  tft.setFont(&FreeSans12pt7b);
  tft.setCursor(10, 180);
  tft.print("Set Point: ");
  tft.setFont(&FreeSans18pt7b);
  tft.print(setPoint);
  tft.setFont(&FreeSans12pt7b);
  tft.print("  KPa");
  tft.setFont(&FreeSans9pt7b);
  tft.setCursor(360, 20);
  tft.print("Relay Status"); 

  //Set-up Timer1 for a 1 sec interrupt
  noInterrupts();           // disable all interrupts
  TCCR1A = 0;
  TCCR1B = 0;
  TCNT1  = 0;

  OCR1A = 62500;            // compare match register 16MHz/256/2Hz
  TCCR1B |= (1 << WGM12);   // CTC mode
  TCCR1B |= (1 << CS12);    // 256 prescaler 
  TIMSK1 |= (1 << OCIE1A);  // enable timer compare interrupt
  interrupts();             // enable all interrupts
}

void loop(){}`

pljakobs, 2018-03-07:
if you just try the 'Draw static text' part, are you getting the same problem? 
What you are saying is, that after 
```
  tft.setFont();
  tft.setCursor(5,30);
  tft.setTextSize(15);
  tft.print("000"); //tft.print(pressure);
```
you still get output in the "default" font or none at all?

Also, this is exactly the use case I started my Gauges library for, you may want to have a look at https://github.com/pljakobs/Arduino_Gauges 
Although, on an Uno, with it's limited memory resources, you won't be getting the best results since it cannot do off-screen drawing. There are still issues, too.

pj

cmmonaco, 2018-03-07:
It prints everything correctly, including the "000", however, all following setFont commands appear to be ignored and the text is printed as absurdly large gfxfont text.

cmmonaco, 2018-03-08:
That fixed it. I just have to apply a setTextSize(1)  before making future setFont calls. Thanks for the help, I doubt I'd ever figure that out on my own!

closed 2018-03-08 by cmmonaco

pljakobs, 2018-03-09:
@PaintYourDragon why are we honoring textsize for bitmap fonts anyway?
Ok, it's technically a way to scale them while not using the extra memory for another, double size (and thus 4x mem footprint) font, but who uses that?
I suggest we remove that for the bitmap fonts

cmmonaco, 2018-03-09:
I would second that. The only reason I didn't think to put in a setFont(1) was because I didn't think it applied to the bitmap fonts.