adafruit/Adafruit-GFX-Library #154: tft.println missing in my library

smit-github, 2018-03-16:
Dear All,

I am using the GLX library for the TFT display of adafruit based TF display shield..

I am going through the library but i cant fine th println function in the Adafruit_GLX.cpp and its .h file can anyone find the function of println under that library..

pljakobs, 2018-03-16:
are you saying that you cannot build your project because you're getting an error that tft.println cannot be found?
println is not a part of Adafruit-GFX but of the underlying core, this library just extends on it. (I think it might be stream.h or so).
Some of the smaller cores (the original Arduino and the Tiny ones I know) do not provide println, so you will have to make do with individual print statements.
The cores for the more powerful controllers (Teensy, ESP8266 etc) provide println and it is then also available for Adafruit-GFX

smit-github, 2018-03-16:
See to be honest i want to learn how you print the string on the display by having println function i didnt found any println function under GFX file so i have where should i found out the same Can you please help me for the same @pljakobs

pljakobs, 2018-03-17:
@smit-github what platform are you on? Again, as I said above, if you use an Atmel based Arduino, you won't have printf, you'll have to use a series of "normal" print statements instead.

ZinggJM, 2018-03-18:
@smit-github you asked the same question in the Arduino Forum Displays.
Adafruit_GFX inherits from Print, as you have already been told.
You can find Print.h in the platform dependent package of your Arduino IDE installation, e.g. in
C:\Users\xxx\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\Arduino on Windows.
For some other packages you find it in e.g.
C:\Users\xxx\Documents\Arduino\hardware\espressif\esp32\cores\esp32
You find there many more interesting files to look at and learn from.
Good Luck!

ZinggJM, 2018-03-18:
@pljakobs I did not find how to open an issue in your Adafruit_ftGFX library. There is Open_Sans_Bold_18pt.h missing in the master branch. What are your plans with this library? Thank you.

pljakobs, 2018-03-18:
@ZinggJM the fonts that are provided in Adafruit-GFX now have been derived from my ftGFX library which has since (for almost three years I think?) not been maintained. You're best off using the current Adafruit-GFX code.

ZinggJM, 2018-03-18:
@pljakobs Seems I am confused. I see no setFont(uint8_t f); method in Adafruit_GFX.h, and had crash when passing NULL (need to re-check).

pljakobs, 2018-03-18:
@ZinggJM did you use the version checked out from this repo?

It's right here:
```
void Adafruit_GFX::setFont(const GFXfont *f) {
    if(f) {            // Font struct pointer passed in?
        if(!gfxFont) { // And no current font struct?
            // Switching from classic to new font behavior.
            // Move cursor pos down 6 pixels so it's on baseline.
            cursor_y += 6;
        }
    } else if(gfxFont) { // NULL passed.  Current font struct defined?
        // Switching from new to classic font behavior.
        // Move cursor pos up 6 pixels so it's at top-left of char.
        cursor_y -= 6;
    }
    gfxFont = (GFXfont *)f;
}
```

ZinggJM, 2018-03-18:
@pljakobs Yes, I use the master branch version of this repository.
Passing NULL works fine (don't know why I thought it didn't).
I thought there were several "classic" fonts, that's why I got confused.
And I experiment with different libraries searching for additional fonts and font rendering.
I am looking forward for the next release of Adafruit_GFX; I like it, gives solid ground for libraries that derive from it. Thank You.

pljakobs, 2018-03-18:
@ZinggJM let me set one one thing straight: I am by no means a main contributor to Adafruit-GFX, I just did the groundwork for the fonts rendering stuff. There are others like @ladyada and @PaintYourDragon who did (and still do) a whole lot more to make this what it is today.
On the fonts: if you are working with the current master branch, then you should have a folder "Fonts".
I'm not quite sure what you are trying to do, but you can always include the .h files from the fonts and then pass the pointer to the structure therein to setFont().
```
#include <FreeMonoOblique9pt7b.h>
...
setFont(&FreeMonoOblique9pt7b);
```
But I do assume you are aware...

ladyada, 2018-03-18:
println() is provided by Stream, we dont have to implement it :)

closed 2018-03-18 by ladyada

ZinggJM, 2018-03-18:
@pljakobs Yes, that's what I use in my examples for my GxEPD library for e-paper displays from Dalian Good Display and Waveshare. Some users of my library asked for bigger fonts and Umlauts. But for fonts I am a total Newbie, my library relies on Adafruit_GFX for graphics and font rendering. For TFTs I had used several different libraries, some had interesting additional fonts.