adafruit/Adafruit-GFX-Library #155: VS Code cannot find include file in browse.path

CoderNM, 2018-04-01:
Using Visual Studio Code v1.21.1
Windows10
Arduino Zero

VS Code cannot find the include <adafruit_GFX.h>
I have the includePath and browse path in the c_cpp_properties.json

what interesting is that VS Code finds the Adafruit_ILI9341.h but can't find Adafruit_GFX.h

Anyone seen this issue or have any suggestion to solve? 

Just for reference, here is the c_cpp_properties.json file,
'''{
    "configurations": [
        {
            "name": "Mac",
            "includePath": [
                "/usr/include",
                "/usr/local/include",
                "${workspaceFolder}"
            ],
            "defines": [],
            "intelliSenseMode": "clang-x64",
            "browse": {
                "path": [
                    "/usr/include",
                    "/usr/local/include",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": true,
                "databaseFilename": ""
            },
            "macFrameworkPath": [
                "/System/Library/Frameworks",
                "/Library/Frameworks"
            ]
        },
        {
            "name": "Linux",
            "includePath": [
                "/usr/include",
                "/usr/local/include",
                "${workspaceFolder}"
            ],
            "defines": [],
            "intelliSenseMode": "clang-x64",
            "browse": {
                "path": [
                    "/usr/include",
                    "/usr/local/include",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": true,
                "databaseFilename": ""
            }
        },
        {
            "name": "Win32",
            "includePath": [
                "${workspaceFolder}",
                "C:/Program Files (x86)/Windows Kits/8.1/Include/um",
                "C:/Program Files (x86)/Windows Kits/8.1/Include/shared",
                "C:/Program Files (x86)/Windows Kits/8.1/Include/winrt",
                "C:\\Users\\Greg\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.6.18\\cores\\arduino",
                "C:\\Users\\Greg\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.6.18\\cores\\arduino",
                "C:/Program Files (x86)/Arduino/libraries/Adafruit_ILI9341",
                "C:/Program Files (x86)/Arduino/libraries/Adafruit_GFX"
            ],
            "defines": [
                "_DEBUG",
                "UNICODE",
                "_UNICODE",
                "<var>"
            ],
            "intelliSenseMode": "msvc-x64",
            "browse": {
                "path": [
                    "${workspaceFolder}",
                    "C:/Program Files (x86)/Windows Kits/8.1/Include/um",
                    "C:/Program Files (x86)/Windows Kits/8.1/Include/shared",
                    "C:/Program Files (x86)/Windows Kits/8.1/Include/winrt",
                    "C:/Program Files (x86)/Arduino/libraries/Adafruit_AM2315",
                    "C:\\Users\\Greg\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.6.18\\cores\\arduino",
                    "C:/Program Files (x86)/Arduino/libraries/Adafruit_ILI9341",
                    "C:/Program Files (x86)/Arduino/libraries/Adafruit_GFX"   
                ],
                "limitSymbolsToIncludedHeaders": false,
                "databaseFilename": ""
            },
            "cStandard": "c11",
            "cppStandard": "c++17"
        }
    ],
    "version": 3
}'''

BMoinak, 2018-04-01:
Get the library of adafruit_GFX.h

CoderNM, 2018-04-01:
I've got the adafruit_GFX library located as shown below,
C:/Program Files (x86)/Arduino/libraries/Adafruit_GFX

But for some reason, Visual Studio Code can't find it.

BMoinak, 2018-04-01:
Check with the arduino IDE to see if it is correctly added to the location.

per1234, 2018-04-01:
>VS Code cannot find the include <adafruit_GFX.h>

The filename is Adafruit_GFX.h, not adafruit_GFX.h. While Windows is not filename case sensitive, VS Code may be. I know the Arduino IDE is filename case sensitive for dependency resolution, even on Windows.

CoderNM, 2018-04-08:
Thank you for the replys.  I could not get VS Code to build a Arduino Sketch that had a Adafruit_GFX library,  (it does build the blink sketch) so I'm trying Atmel Studio.  Not that much different, but gotta try something different.

Now, in Atmel Studio, I get the following errors with Adafruit_GFX libraries, 

recipe for target 'src/libraries/adaftuit-gfx/Adafruit_SPITFT.o' failed
'SPE' was not declared in this scope
'_BV' was not declared in this scope
etc, etc, etc.

Has anyone seen this error or know how to fix?

I do have an open support case with Microchip, so when/if it gets solved, I'll be sure to post the fix.

strech345, 2019-06-26:
same Problem,

have pio 4.0.0b3 installed after 3 was also not finding the source files.
I also change C_Cpp: Intelli Sense Engine to Tag Parser. Also not working.

Now after some playing with this settings i don't get any error at main.cpp
But on compiling i get the following error: 

```
Compiling .pio\build\esp32dev\src\main.cpp.o
Generating partitions .pio\build\esp32dev\partitions.bin
Compiling .pio\build\esp32dev\lib8ce\Wire\Wire.cpp.o
Compiling .pio\build\esp32dev\lib5e1\SPI\SPI.cpp.o
Compiling .pio\build\esp32dev\libcdd\Adafruit SSD1306_ID135\Adafruit_SSD1306.cpp.o
.pio\libdeps\esp32dev\Adafruit SSD1306_ID135\Adafruit_SSD1306.cpp:51:26: fatal error: Adafruit_GFX.h: No such file or directory
```

Does someonknow how to use it right on platformio for esp32?

strech345, 2019-06-27:
Maybe i found the mistake...
the lib is not `Adafruit_GFX` it is `Adafruit GFX Library`

caternuson, 2020-04-06:
Closing. Not a library issue.

closed 2020-04-06 by caternuson