adafruit/Adafruit-GFX-Library #157: Default font looks better with no serifs on 'T'

bperrybap, 2018-04-07:
The default font in glcdfont.c is using serifs on the capital T 
it is single pixel hanging one pixel down on each end of the horizontal line.

When animated (moving horizontally) on led panels, it looks like the top line is two pixels wide rather than a single like with serifs because the font is so narrow and a bit of persistence of vision.
It looks much better when the serifs are removed.
No other character is using serifs.
Here are the codes for 'T"
```
//	0x03, 0x01, 0x7F, 0x01, 0x03, // 0x54 T (with serif)
	0x01, 0x01, 0x7F, 0x01, 0x01, // 0x54 T (without serif)
```

mikeysklar, 2026-10-29:
Changing glyphs in the default font would change how existing sketches look.

The README puts backward compatibility first, so the font stays as it is.

You can edit the 'T' in `glcdfont.c` in your own copy of the library.

Closing this one.

closed 2026-10-29 by mikeysklar

bperrybap, 2026-10-29:
Sure users can edit their personal copy of the code, but that doesn't really solve the issue.
Keep in mind no other character in the font is using serifs. The T is actually the exception rather than the rule for this font.
I think a better alternative than just blowing this off, would be add a comment line so those that may run into this can look at the source code and simply un-comment the line without the serif and comment out the default line with the serif.
i.e. provide a bit of help for those that run into this issue on LED based scrolling devices.

Maybe change the current line in the code to these lines:

```
	0x03, 0x01, 0x7F, 0x01, 0x03, // 0x54 T (with serif)
//	0x01, 0x01, 0x7F, 0x01, 0x01, // 0x54 T (without serif - which may look better on LED type displays)
```
This gives users an easy way to see how to change the font to get rid of the serif on the T character.

bperrybap, 2026-10-29:
I can do a pull request if you will accept it.

mikeysklar, 2026-10-29:
Sure. A comment-only change would be fine. I'll accept a PR.