adafruit/Adafruit-GFX-Library #158: ESP32 unable to initialize wifi. (serial monitor show connecting........)

amjrazman, 2018-04-20:
Hi .

Im using ESP32 for the module. I want to connect it with wifi. however ,  on serial monitor it shows (connecting .............................) till forever. I wait for hours tho i believe it will connect somehow. 
Anyway im using MacBook Pro, run with El Capitan OSX. 

Below is the code: 

*************************************************************
`#include <WiFi.h>
char* ssid     = "ooooooooo";
const char* password = "xxxxxxxxxx";
WiFiServer server(80);

 /*MLX90614 Sensor */
#include <Wire.h>
#include <Adafruit_MLX90614.h>

Adafruit_MLX90614 mlx = Adafruit_MLX90614();
 
void setup() 
{
  Serial.begin(115200);
  delay(10);
  Serial.println("Adafruit MLX90614 test");  
    //pinMode(21,INPUT)
    //adc1_config_width(ADC_WIDTH_10Bit);
 //   adc1_config_channel_atten(ADC1_CHANNEL_5,ADC_ATTEN_11db);
//  duration=1000*10;
 //time1=0;
  mlx.begin(); 

  connectWiFi();
}

void loop() 
{
   WiFiLocalWebPageCtrl();
 /* 
  Serial.print("Ambient = "); 
  Serial.print(mlx.readAmbientTempC()); 
  Serial.print("*C\tObject = "); 
  Serial.print(mlx.readObjectTempC()); 
  Serial.println("*C");
  Serial.print("Ambient = "); 
  Serial.print(mlx.readAmbientTempF()); 
  Serial.print("*F\tObject = "); 
  Serial.print(mlx.readObjectTempF()); 
  Serial.println("*F");
 
  Serial.println();
  delay(500);
 */ 
}

/***************************************************
* Send and receive data from Local Page
****************************************************/
void WiFiLocalWebPageCtrl(void)
{
  WiFiClient client = server.available();   // listen for incoming clients
  //client = server.available();
  if (client) {                             // if you get a client,
    Serial.println("New Client.");           // print a message out the serial port
    String currentLine = "";                // make a String to hold incoming data from the client
    while (client.connected()) {            // loop while the client's connected
      if (client.available()) {             // if there's bytes to read from the client,
        char c = client.read();             // read a byte, then
        Serial.write(c);                    // print it out the serial monitor
        if (c == '\n') {                    // if the byte is a newline character

          // if the current line is blank, you got two newline characters in a row.
          // that's the end of the client HTTP request, so send a response:
          if (currentLine.length() == 0) {
            // HTTP headers always start with a response code (e.g. HTTP/1.1 200 OK)
            // and a content-type so the client knows what's coming, then a blank line:
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();

            // the content of the HTTP response follows the header:
            //WiFiLocalWebPageCtrl(); 
              client.print("Ambient = "); 
              client.print(mlx.readAmbientTempC());
              client.print("*C\tObject = ");
              client.print(mlx.readObjectTempC()); 
              client.print("*C");
              client.print("Ambient = "); 
              client.print("<br>");
              client.print(mlx.readAmbientTempF());
              client.print("*F\tObject = "); 
              client.print(mlx.readObjectTempF());
              client.print("*F");
              client.print("<br>");
              
             
            // The HTTP response ends with another blank line:
            client.println();
            // break out of the while loop:
            break;
          } else {    // if you got a newline, then clear currentLine:
            currentLine = "";
          }
        } else if (c != '\r') {  // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }
/*
        // Check to see if the client request was "GET /H" or "GET /L":
        if (currentLine.endsWith("GET /H")) {
          digitalWrite(LED_PIN, HIGH);               // GET /H turns the LED on
        }
        if (currentLine.endsWith("GET /L")) {
          digitalWrite(LED_PIN, LOW);                // GET /L turns the LED off
        }*/
      }
    }
    // close the connection:
    client.stop();
    Serial.println("Client Disconnected.");
  }
}

/***************************************************
* Connecting to a WiFi network
**************************************************/
void connectWiFi(void)
{
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
 // if(WiFi.getAutoReconnect()) {
  WiFi.disconnect();
  WiFi.begin(ssid, password);
 // }
  //else
  //{
    //WiFiSTAClass::_setStatus(WL_CONNECT_FAILED)
  //}
  while (WiFi.status() != WL_CONNECTED) 
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected.");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  //delay(5000);
 server.begin();
} 
`********************************************************************

pljakobs, 2018-04-20:
this is the issues page for the Adafruit-GFX-library.
Your code does not even include that library, so I'm pretty certain that you're asking the wrong people.

kind regards

pj

x4m3, 2018-05-21:
yeah, it's not an issue with the GFX-library, you must have a problem with your ESP32.

make sure your wifi network is up and running, and restart your esp a bunch of times if it doesnt connect -- on my board the wifi works after 5 reboots or something.

good luck!

pljakobs, 2018-07-16:
@ladyada I think this is a candidate for closing, too.

closed 2018-07-16 by ladyada