adafruit/Adafruit-GFX-Library #160: wiznet5100 ethernet and led matrix

kessero, 2018-05-25:
My project assumes creating server with Ethernet shield w5100 and matrix led 16x32 with adafruit GFX library. When I include <Ethernet.h> led matrix stops working. It seems that there's a conflict. Anybody can help me with that?

pljakobs, 2018-05-26:
First: you probably should ask this question in one of the arduino forums. This is the issues page for the GFX Library and should really only deal with bugs found in the library itself.
That said: I would assume that your problem is that both pieces of hardware use the same internal hardware / pins and collide over that. I have not looked at the devices you describe, but I assume they either both use I2C or SPI. For I2C to work, the devices must have different addresses and the termination must be correct, for SPI, the devices must have different CS pins and those pins must be configured correctly when setting up the device.
But as said: please take it to the correct (hardware) forums.

kessero, 2018-06-16:
Problem solved. I made my project on arduino MEGA. Problem was in memory.

pljakobs, 2018-07-16:
@ladyada and another one

ladyada, 2018-07-16:
zoom!

closed 2018-07-16 by ladyada