adafruit/Adafruit-GFX-Library #161: Please provide very small font sizes in default repository.

baryluk, 2018-06-03:
Hi,

It would be nice to include few smaller sizes

3x3
4x3
5x3
5x4

(the glyph sizes, no margin included). Usage of kerning for some glyphs would also be beneficial, to squeeze max of space.

As well add option to not add interline or spaces between characters. I do have 320x240 LCD display using IDL9340, and the default 8x5 is very nice, but because I use it to display various diagnostic data in real time, having bigger density is a huge plus.

For the smallest fonts (3x3 and 4x3), I also force glyphs to be upper case before passing to print(char) method. But even lower cases are somehow usable. Especially if user knows what is on display in terms of text (just some labels and units that do not change). The numbers are also distinct and readable enough.

Another option to research is to have grayscale fonts, instead of just bitmap, as usage of additional color gradations could improve some small fonts (by fuzzing some corners similar to subpixel rendering). They can be all pre-computed offline too.

I know I could do that all myself, but I am certain many other people would benefit of having this in official repo :)

myhd, 2018-09-21:
I think a 3x3 pixel font will be rather useless.

pljakobs, 2018-09-24:
[this](https://www.deviantart.com/quanyails/art/Smallest-Font-Minifont-157896188) is a 6x4 (or 6x3 if you don't count the spacing) and I think it's already very hard to read. Not sure if that kind of font really makes any sense at all.
It shouldn't be a huge amount of work, though, to create the corresponding tables manually. 
I wonder, though, picopixel already provides a 6 pixel height font - I guess that would be the same thing.

Megawaffles, 2019-03-08:
> 
> 
> Nevermind, I found 3x3 and 5x3 in default repo. They work nice for me.

Where did you find this? I'm trying to drive a matrix and the text is a bit too large.

sergiitk, 2019-06-26:
They're in [`Fonts/`](https://github.com/adafruit/Adafruit-GFX-Library/tree/master/Fonts), just labeled by their name. As of June 2019, here's what's available:
- Org_01.h
- Picopixel.h
- Tiny3x3a2pt7b.h
- TomThumb.h

StefanL38, 2020-04-03:
There is a solution for a lot of fonts.
On the website http://oleddisplay.squix.ch/#/home

You can choose from fonts and can choose the **SIZE** of the font and it will create the header-file!
Then you can copy & paste this headerfile into a texteditor and save it with .h to the 
C:\Users\username\Documents\Arduino\libraries\Adafruit_GFX_Library\Fonts-Folder

Please donate Daniel Eichhorn  the author of this service at least 1 dollar for a beer.
https://www.paypal.me/squix78

cheers Daniel Great work!

mikeysklar, 2026-11-12:
Four small fonts ship in `src/Fonts/`:

| Font | `A` glyph (w x h) | `ABC xyz 0123456789 -.%` width |
|---|---|---|
| `Tiny3x3a2pt7b` | 3 x 3 | 82 px |
| `TomThumb` | 3 x 5 | 78 px |
| `Picopixel` | 3 x 5 | 78 px |
| `Org_01` | 5 x 5 | 119 px |

```cpp
#include <Fonts/TomThumb.h>
tft.setFont(&TomThumb);
```

Size 1 on the left, size 3 on the right, Feather HUZZAH ESP8266 + 3.5" HX8357 FeatherWing:

[image]

For other sizes, the browser version of fontconvert makes a header from any TTF: https://adafruit.github.io/Adafruit-GFX-Library/fontconvert/

Closing, please reopen if you need something these don't cover.

closed 2026-11-12 by mikeysklar