adafruit/Adafruit-GFX-Library #162: rendering Katakana, Hiragana and Japanese Kanji characters

matphillips, 2018-06-27:
I'm trying to output the full range of Japanese characters to a HUB75 display connected to an ESP32 with no success.

The included font package seems not to include them and from reading the source code for the custom font converter, it seems that this may not be possible.

Is there any way to use this library in this way?

matphillips, 2018-06-27:
Thanks for the reply, good to get confirmation that it's not supported.

You're completely correct about the font file size, but I was hoping the glyphs could be loaded from spiffs as needed or something. Trying to generate a "word of the day" style learning tool to show random words changing each hour, so speed isn't an issue.

I'll keep looking for a solution, thanks!

closed 2018-06-27 by matphillips

pljakobs, 2018-06-27:
we don't support loading fonts from spiffs, the core focus of this library is the Arduino platform as a whole. It has to work from a humble Atmel 8-Bit all the way to ESPs and ARM cores.

hikari-no-yume, 2020-09-06:
In case someone in future is wondering about Japanese with this library.

While it does sound like full Japanese script support is out of scope, I want to point out that there are single-byte character sets for Japanese containing just half-width Katakana (JIS X 0201) and a variant of ASCII (more accurately ISO 646) which are commonly used on dot matrix LCDs (in fact it's sometimes the default character set). I imagine that is something this library could reasonably support?