adafruit/Adafruit-GFX-Library #163: Most Pleasing Font

Dancopy, 2018-07-07:
Greetings!
How can I edit this Font to make it nicer or fatter? Thank you

pljakobs, 2018-07-07:
If you build the makefont tool for your platform (easy on Linux and someone has done it on windows, too) you can easily convert any true type font in the size you like.
You could theoretically edit the font files, but there are no tools for the format and it will be next to impossible to edit the packed format manually. 
Fontconvert is your friend.

mikeysklar, 2026-10-21:
fontconvert turns any TrueType file into a font at the size you choose.
For a heavier look, pick a bolder TTF, since fontconvert keeps the weight of the source.
Usage: fontconvert fontfile size [first] [last], see fontconvert/fontconvert.c:72.
Closing this one.

closed 2026-10-21 by mikeysklar