adafruit/Adafruit-GFX-Library #164: Overflow in filled triangle routine

static-void, 2018-07-13:
I have found a case that causes overflow in the triangle routine, which also causes massive graphical glitches. My naive solution is to simply use int32_t for all the locals, but there is almost certainly a better way (I don't have time to think on it).

The counter example is x0 = 314, y0 = 11, x1 = 14, y1 = 200, x2 = 97, y2 = 237

pljakobs, 2018-07-13:
without having tested it, I would assume that this is not so much about the sizeof(coordinates) but about their of-screenness. I assume that at least some of your corners are off screen?
All coordinates are int16_t and the routine wouldn't fare well if one wraps around. 
What was the display resolution you tested this on?

static-void, 2018-07-13:
It's a 320x240 screen (WxH) so all coordinates are on screen. I think an intermediate value overflows in the first loop. I end up with a value for the variable a (IIRC) of 400-something which is off screen (and wrong) and then the drawing goes wrong.

static-void, 2018-07-13:
This is on a cortex m4 device and gcc 6.3.1 at O2, which may or may not be relevant. If you like I could produce an assembly dump of the function.

pljakobs, 2018-07-13:
I currently have no platform to reproduce this - could you try digging into the error a bit more?
Also, I have really never looked at the triangles, I just happen to be working on some other GFX code today, hence I responded ;-)

static-void, 2018-07-13:
I am not sure if the bug is platform dependent or not. I will investigate more but not immediately, hopefully I can catch it in the debugger. The bug must be subtle to have been missed for so long. It's also hard to see how it could overflow since there is no multiplication, but I am making that assumption from what I've seen so far (a result was larger than it should be, and the problem goes away if you make the local variables wider).

pljakobs, 2018-07-13:
true, I was briefly thinking about a /0, but that's not the case either.
When you say it's an Arm m4, what board is that? I mostly use Teensy 3.x es and have not seen the issue before. Albeit triangles aren't my usual thing ;-)

static-void, 2018-07-13:
It's not an Arduino system, but a custom board with a SAM4N16B running FreeRTOS and an ILI9341 connected via SPI.

pljakobs, 2018-07-31:
I have finally managed to test your data on a teensy 3.2 - and I'm not getting any strange effects. Not sure what issue you're seeing.

pj

mikeysklar, 2026-10-29:
Tested on current master with a 320x240 canvas.
The sa/sb accumulators have been int32_t since 2014.
That is the same fix suggested here.

| test | mismatched pixels |
|---|---|
| your triangle (314,11 14,200 97,237) | 0 |
| 20,000 random triangles | 0 |
| same code with int16 accumulators | 20,284 |

16-bit int (AVR) emulation also gives 0.
Closing this one.
Please reopen if you see it on the current release.

closed 2026-10-29 by mikeysklar