adafruit/Adafruit-GFX-Library #165: setCursor does not set start position for println

voxspox, 2018-07-16:
Adafruit_GFX: 1.2.5
Arduino IDE: 1.8.5
Arduino Uno Board

`setCursor()` does not set the proper start position for `println()`. If I do not use setCursor several println calls will print off the bottom of the screen. If I call p.ex. `setCursor(3, 3)` the text will start at (0, 0) but not at (3, 3)

pljakobs, 2018-07-16:
@voxspox is that using the "new" proportional fonts?
If so, it's by design. For the  proportional fonts, the "cursor" marks the baseline (the bottom line of the character bodies) and the actual character sits on top of that as high as it needs to and, if it has an understroke, it will reach below the baseline. 
If you want to position a string with proportional fonts correctly, use the charBounds() or getTextBounds() functions.
Alternatively, you might try to have a look at the Gauges lib I wrote, it allows to place text boxes on screen and then update them as needed. https://github.com/pljakobs/Arduino_Gauges (beware, there are still bugs, especially in the tape gauges)

ladyada, 2018-07-16:
closing since theres an answer - re-open if that's not the solution :)

closed 2018-07-16 by ladyada