adafruit/Adafruit-GFX-Library #166: future development

pljakobs, 2018-07-24:
I'm opening this as an issue since it seems to be the way to reach out to other developers to this library.

We've been functionally mostly stuck ever since we've added font support a few years ago. some of my develoment hit the limits of what this library can do, especially with the limited available RAM on some platforms (even on the Teensy!) because I make a lot of use of canvas objects.
I have, over the last months, driven two major projects:
- a seemless 24 Bit interface that allows to decouple applications from the underlying hardware bit depth (that is: 16 Bit applications should continue to work even if the hardware supports 24 bits and 24 bit applications should work without having to know if the hardwre supports 24 bit or not)
- a bitmapped canvas object. This is really inspired from the work I did decades back on the Amiga. GFXiCanvas allows to define a canvas that is really composed of a number of GFXcanvas1 single bit objects. Up to eight bits are allowed and thus up to 256 colors are possible. This, of course, requires a color palette, just as back in the Amiga days, and the palette can be defined using RGB24 values.

At this point, I have a prototype that works, albeit very rough, for setting color dots and displaying them. 
If you want to help work out the bugs or help optimize the code (it's pretty slow and non optimized right now), have a look at https://github.com/pljakobs/Adafruit-GFX-Library/tree/Feature-bitplanes

thx 

pj

jamesdanielv, 2018-08-24:
hello. would like to suggest code change that i am currently using with my custom build. this is my second posting of this code, originally posted to st77xx.cpp area, and it was closed and i was asked to put it into this section. seems code is migrating to this file.
what i hope happens is the value is seen in this and the overhead it reduces. it reduces the setwindow and address call to one time per 4 pixel writes as long as pixel data is known somewhere in advance. how that happens is creative of the user.

also if this works well, i have a 16x speed transfer as well

it is unrolled a bit so some unconventional steps are used to make it more manageable, such as '#define' used to create complex commands. also spi commands are unrolled and directly places as timing can be critical. have included safeguards for proper operation even at lower spi speeds that ensure spi is ready.

this code takes a large rectangle area, and subdivides it into 4 areas that can have different color values. while(!(SPSR & (1<<SPIF) )) is used as a safety check but should normally run thru without loop back or jmp performance tax, unless spi is at a lower speed. within this function spi is burst at about 80-100% increase speed from normal at max spi data rate, and overhead of pixel location and window size is reduced by 4x!

void Adafruit_ST77xx::fillRectFast4colors(int16_t x, int16_t y, int16_t w, int16_t h,
  uint16_t color0,uint16_t color1,uint16_t color2,uint16_t color3) {
//this is needed for text but not fills with solid color!
  // rudimentary clipping (drawChar w/big text requires this)
 // if((x >= _width) || (y >= _height)) return;
 // if((x + w - 1) >= _width)  w = _width  - x;
 // if((y + h - 1) >= _height) h = _height - y;

  setAddrWindow(x, y, x+w-1, y+h-1);

  SPI_BEGIN_TRANSACTION();
  //created complex commands in #define so code can be a lot cleaner
  //this is a complex command that complexspi= hi[0] value
#define complexspi SPCRbackup = SPCR; SPCR = mySPCR; SPDR 
//this is small delay
#define shortSpiDelay  __asm__("nop\n\t");// __asm__("nop\n\t");// __asm__("nop\n\t");
//last comment out ';' not needed here, but can be added in commands later 
#define longSpiDelay __asm__("nop\n\t"); __asm__("nop\n\t");__asm__("nop\n\t"); __asm__("nop\n\t");  __asm__("nop\n\t"); __asm__("nop\n\t"); __asm__("nop\n\t"); __asm__("nop\n\t"); __asm__("nop\n\t"); __asm__("nop\n\t"); __asm__("nop\n\t");// __asm__("nop\n\t");// __asm__("nop\n\t");// __asm__("nop\n\t");// __asm__("nop\n\t")  
  
digitalWriteFast(TFT_DC,HIGH);// old DC_HIGH();
digitalWriteFast(TFT_CS,LOW);// old CS_LOW();

//  y=h;
  uint16_t countx;//we use these to understand what to draw!
  uint16_t county;

 SPCR = SPCRbackup;  //we place at top for next loop iteration
county=h/2;

while (county !=0 ){countx=w/2;while (countx !=0) {while(!(SPSR & (1<<SPIF) ));;complexspi = highByte(color0);longSpiDelay;while(!(SPSR & (1<<SPIF) ));
complexspi =lowByte(color0) ;shortSpiDelay;countx--;}
while(!(SPSR & (1<<SPIF) ));

countx=w/2; while (countx !=0) {while(!(SPSR & (1<<SPIF) ));;complexspi =highByte(color1);longSpiDelay;while(!(SPSR & (1<<SPIF) ));
complexspi=lowByte(color1);shortSpiDelay;countx--;}
while(!(SPSR & (1<<SPIF) ));

county--; }//end of first half of rectange [0][1] drawn!
county=h/2;
while (county !=0){countx=w/2;while (countx !=0) {while(!(SPSR & (1<<SPIF) ));;complexspi = highByte(color2);longSpiDelay;while(!(SPSR & (1<<SPIF) ));
complexspi =lowByte(color2);shortSpiDelay;countx--;}
while(!(SPSR & (1<<SPIF) ));

countx=w/2;while (countx !=0) {while(!(SPSR & (1<<SPIF) ));;complexspi = highByte(color3);longSpiDelay;while(!(SPSR & (1<<SPIF) ));
complexspi =lowByte(color3);shortSpiDelay; countx--;}
county--; }//end of first half of rectange [2][3] drawn!
while(!(SPSR & (1<<SPIF) ));
digitalWriteFast(TFT_DC,HIGH);// old DC_HIGH();
digitalWriteFast(TFT_CS,HIGH);//old CS_HIGH();
SPI_END_TRANSACTION();
}

mikeysklar, 2026-10-21:
Thanks for sharing this, @pljakobs.

The 24-bit and palette canvas work is not in the library today.

A focused pull request would be welcome for any single piece.

Closing this one, since it is a design discussion rather than a bug.

closed 2026-10-21 by mikeysklar