adafruit/Adafruit-GFX-Library #167: Compile errors  USE_FAST_PINIO is not defined.... And compile on OpenCM and OpenCR boards

KurtE, 2018-07-26:
First part of this issue, is there are compile issues if you try to undefine USE_FAST_PINIO.   
You need ;'s after the digitalWrite commands in the macros.  Also in your header file in the 
#else clause for which board is used, the source file uses #undefine where it should be #undef

I found these trying to get the code to build for a couple of the Robotis boards:
OpenCR1.0 and the OpenCM9.04 boards.  

I was able to make it build using the following changes:

In Adafruit_SPITFT.h:
```
@@ -16,12 +16,14 @@
typedef volatile uint8_t RwReg;
#elif defined(ARDUINO_STM32_FEATHER)
typedef volatile uint32 RwReg;
#elif defined(__OPENCR__) || defined (__OPENCM904__)
#undef USE_FAST_PINIO
#elif defined(ARDUINO_FEATHER52) || defined(__arm__)
typedef volatile uint32_t RwReg;
#elif defined(ESP32) || defined(ESP8266)
typedef volatile uint32_t RwReg;
#else
#undefine USE_FAST_PINIO
#undef USE_FAST_PINIO
#endif

#include "Adafruit_SPITFT_Macros.h"
```

Then fix a few of their issues in Adafruit_SPITFT_Machros.h
```
@@ -13,8 +13,8 @@
#else
#define SPI_DC_HIGH()           digitalWrite(_dc, HIGH)
#define SPI_DC_LOW()            digitalWrite(_dc, LOW)
#define SPI_CS_HIGH()           { if(_cs >= 0) digitalWrite(_cs, HIGH) }
#define SPI_CS_LOW()            { if(_cs >= 0) digitalWrite(_cs, LOW)  }
#define SPI_CS_HIGH()           { if(_cs >= 0) digitalWrite(_cs, HIGH); }
#define SPI_CS_LOW()            { if(_cs >= 0) digitalWrite(_cs, LOW);  }
#endif

```
As these boards do not have the same concept of being able to write a bit to a specific memory location to turn on or off an IO pin.

Would be great if you added this to your sources.

Thanks

ladyada, 2018-07-26:
thanks! you can submit a PR and we'll take a look!

KurtE, 2018-07-26:
created pull request: 
https://github.com/adafruit/Adafruit-GFX-Library/pull/168

closed 2018-07-26 by ladyada