adafruit/Adafruit-GFX-Library #177: description for the canvar functionality?

breezer15, 2018-09-12:


pljakobs, 2018-09-12:
may I suggest you take a look at https://github.com/pljakobs/Arduino_Gauges ? I wrote the Gauges class to do just that: display data without flicker and actually without having to do a whole lot of code yourself.
It's by far not perfect yet, but it does work. 
At the very least, you should get a good idea of how to use GFXcanvas16 (and GFXcanvas1 doesn't work any different) by looking at the textGauge::redraw() member function.
Please be aware that this version needs a slightly extended version of AdafruitGFX, since it is written to be display technology agnostic and on OLEDs, we need to call display.display() after updating the display content and that is a function that is not generally in AdafruitGFX but only in OLED subclasses.
Here's an AdafruitGFX with display() as a virtual function.
It has to be said that the Gauges library is by no means done yet, but text and (basic) bar graphs work well, provided you have enough RAM for the amount of off-screen buffer you need.
If you feel a little more adventurous, you may want to look at my highly experimental, yet working, extension to [AdafruitGFX](https://github.com/pljakobs/Adafruit-GFX-Library/tree/Feature-24Bit) with variable depth canvas objects called GFXicanvas. (see here: https://github.com/pljakobs/Adafruit-GFX-Library/tree/Feature-24Bit)
This, besides enabling a 24Bit color model, adds a canvas object that can have anywhere from 1 to 8 bits (and uses GFXcanvas1 to implement the required number of bitplanes) and is intended as a companion to the gauges library in order to provide for a memory conserving model.
Again, at the very least, it should show you how to use GFXcanvas1 objects.
(hint: create them and use them just as you would use any display object)

pj

pljakobs, 2018-09-15:
it would be helpful to get feedback from people who do open cases.

breezer15, 2018-09-17:
Yes you are right, thank you for your solution. Now i can create and use RAM resident canvases. With this solution it is possible to update custom fonts without flickering: 

`_canvas = new GFXcanvas1(w, h);`
`_canvas->fillScreen(BLACK);`
`_canvas->setFont(&FreeMonoBold12pt7b);`
`_canvas->setTextColor(WHITE, BLACK);`
`_canvas->setTextSize(fontsize);`
`_canvas->setCursor(x, y);`
`_canvas->print(strg);`
`tft.drawBitmap(x-(x, y,_canvas->getBuffer(), w, h, WHITE, BLACK);`
`delete _canvas;`

closed 2018-09-17 by breezer15

pljakobs, 2018-09-17:
Thanks for the response.

it should be noted that, depending on your system, it might be beneficial to not delete the canvas and recreate it with every redraw (even though that's what the gauges library does). The risk with that is that you fragment your RAM and, at runtime, run out of available memory.
If you can statically allocate that canvas and use it throughout (a 1 bit canvas should be no big issue on most modern platforms), you would probably buy yourself a bit more runtime stabiltity.

pj