adafruit/Adafruit-GFX-Library #180: please separate Adafruit_SPITFT from Adafruit_GFX library

ZinggJM, 2018-09-23:
I used Adafruit_GFX with many different Arduino IDE platforms.
It used to compile for all platforms I used without problems.
But now it no longer compiles for some platforms because of the added Adafruit_SPITFT part.
Platforms affected I encountered so far are for STM32: STM32DUINO and STM32GENERIC.

Adafruit_GFX is the base graphics library used for Adafruit libraries, but also for many other libraries.
Adafruit_SPITFT is a useful addition for some libraries, but should be in its own library.

Thank you for your great work!

ZinggJM, 2018-09-23:
#define RwReg uint32_t
in 
C:\Users\xxx\Documents\Arduino\hardware\STM32GENERIC\STM32\cores\arduino/stm32/stm32_gpio.h:43:15:
conflicts with 
  typedef volatile uint32_t RwReg;
in 
C:\Users\xxx\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_SPITFT.h:23:29:
and
   #include "wiring_private.h"
does not exist for some platform

ladyada, 2018-09-23:
i think this is a pretty minor change, and i'd like to keep SPITFT in here because 99% of GFX users will also use SPITFT - could you submit a PR that fixes this for STM32? we dont own any STM32 boards to test with :)

WereCatf, 2018-09-23:
I cannot find any reference to stm32_gpio.h in the Arduino OTTO - core that's installable straight from the Arduino IDE by default, in Roger Clark's STM32Duino - core or in the ST-official STM32Duino - core, so I can't do anything about that, but fixing the issue with "wiring_private.h" is simple:

In Adafruit_SPITFT.cpp change the line `#ifndef RASPI` to `#if !defined(RASPI) && !defined(STM32F4)` and the compile-error goes away in all the cores I just tested.

ZinggJM, 2018-09-24:
stm32_gpio.h is in https://github.com/danieleff/STM32GENERIC
https://github.com/danieleff/STM32GENERIC/blob/master/STM32/cores/arduino/stm32/stm32_gpio.h
and RwReg is still defined as a macro in the actual version, unfortunately.
So I would propose to replace RwReg by RwReg_t in Adafruit_SPITFT (as it is a typedef).

WereCatf, 2018-09-26:
Did a pull-request; compiles fine, didn't test on actual hardware. Leaving that part to @ZinggJM

ZinggJM, 2020-05-17:
**It got worse lately**

Adafruit_GFX is a base class. As a library it should **not** depend on any other library!

ladyada, 2020-05-17:
@ZinggJM did you try https://github.com/adafruit/Adafruit-GFX-Library/pull/181 ? if you test it working we could merge it in - thanks!

ZinggJM, 2020-05-17:
@ladyada,

no, sorry, I am too busy for this test right now.

My recent comment is not related to this issue.
**I just noticed that I get more and more libraries added with recent Adafruit_GFX updates.**

I think the time has come to fix this issue with Adafruit Libraries structures.

ladyada, 2020-05-17:
why does it matter how many dependacies or objects are in a class? the compilation time is the same, ide handles deps. please respond in a positive, constructive manner. there is a pr to fix the very issue mentioned here

ZinggJM, 2020-05-17:
I don't think the compilation time is the same, as the Arduino IDE searches through all libraries.

I try to make positive suggestions. I don't want to criticize. Separating the library classes would lead to considerably less updates of the Adafruit_GFX library.

closed 2020-05-17 by ZinggJM

ZinggJM, 2020-05-19:
https://forum.arduino.cc/index.php?topic=649404.msg4608135#msg4608135