adafruit/Adafruit-GFX-Library #194: fontconvert set codepage

varna9000, 2019-01-04:
Im trying to pullout chars from codepage iso-8859-5  with the command `fontconvert arial.ttf 12 176 239` but it doesn't pullout correct chars from Arial? I tried to replace the if checks in 'font covert with the said start and end chars, but it still outputting extended ASCII chars. I'm not expert in C, but wouldn't this be an easy thing to do? After all we are limiting the chars, not pulling out the whole utf-8 charset.

Bodmer, 2019-02-09:
I expect the correct glyphs are being pulled from the font but the GFX engine does not handle the UTF-8 encoding and produces garbage output on the screen.

The [version of the library here](https://github.com/Bodmer/Adafruit-GFX-Library) should work. It works with this example:
[hiragana_example.zip](https://github.com/adafruit/Adafruit-GFX-Library/files/2848217/hiragana_example.zip)

 I have generated a pull request.

mikeysklar, 2026-11-12:
fontconvert's first and last are Unicode code points, not codepage bytes. `176 239` pulls U+00B0 to U+00EF (Latin-1), which is why you got extended ASCII.

ISO-8859-5 0xB0 to 0xEF is А to я, which is U+0410 to U+044F. Convert that range and change the font's first/last to the codepage bytes:

1. `fontconvert FreeSerif.ttf 24 1040 1103 > FreeSerifCyrillic24pt.h`
2. In the header, change the last line's `0x410, 0x44F` to `0xB0, 0xEF`.
3. Print ISO-8859-5 bytes:

```cpp
display.setFont(&FreeSerifCyrillic24pt);
display.print("\xBF\xE0\xD8\xD2\xD5\xE2"); // Привет
```

The font has no space, so move the cursor between words. Ё and ё (`\xA1`, `\xF1`) are outside this range. Feather HUZZAH ESP8266 + 3.5" HX8357 FeatherWing:

[image]

Closing, please reopen if this doesn't work with your font.

closed 2026-11-12 by mikeysklar