adafruit/Adafruit-GFX-Library #198: Adafruit_GFX::operator=(const Adafruit_GFX&)' is implicitly deleted

vogt31337, 2019-02-07:
Hi,

**Problem:**
I tried using Adafruit_SSD1306 in a sub class to encapsulate it's behaviour. But Adafruit_GFX resists...
```
void Display::initialize(void) {
    ssd1306 = Adafruit_SSD1306(128, 32, &Wire);
    [..]
}
```

**Versions:**
Arduino ver. 1.8.8
GFX ver. 1.3.6
SSD1306 ver. 1.2.9

**Error:**
```display.cpp: In member function 'void ZamekDisplayClass::initialize()':
display.cpp:4:11: error: use of deleted function 'Adafruit_SSD1306& Adafruit_SSD1306::operator=(const Adafruit_SSD1306&)'

   ssd1306 = Adafruit_SSD1306(128, 32, &Wire);
           ^

In file included from display.h:5:0,
                 from display.cpp:1:

Adafruit_SSD1306.h:115:7: note: 'Adafruit_SSD1306& Adafruit_SSD1306::operator=(const Adafruit_SSD1306&)' is implicitly deleted because the default definition would be ill-formed:

 class Adafruit_SSD1306 : public Adafruit_GFX {
       ^

Adafruit_SSD1306.h:115:7: error: use of deleted function 'Adafruit_GFX& Adafruit_GFX::operator=(const Adafruit_GFX&)'

In file included from display.h:4:0,

                 from zamek-display.cpp:1:

Adafruit_GFX.h:13:7: note: 'Adafruit_GFX& Adafruit_GFX::operator=(const Adafruit_GFX&)' is implicitly deleted because the default definition would be ill-formed:

 class Adafruit_GFX : public Print {
       ^

Adafruit_GFX.h:13:7: error: non-static const member 'const int16_t Adafruit_GFX::WIDTH', can't use default assignment operator

Adafruit_GFX.h:13:7: error: non-static const member 'const int16_t Adafruit_GFX::HEIGHT', can't use default assignment operator
```

**Possible solution:**
Adafruit_GFX.h, line 132: Remove const.

**What's happening?**
As far as I understood this [explanation](https://stackoverflow.com/questions/5966698/error-use-of-deleted-function), you can't assign objects which have const fields in their definition.
My C++ skill is too low to understand the details, but removeing the const expression solves this compilation.

BillyDonahue, 2020-05-28:
It's correct that this object shouldn't be copied around, as it owns resources in a simple raw pointer and implicit way. It's just luck that the `const` is what's enforcing this for you (partially, as it blocks copy-assignment but not copy-construction). You should probably switch to using a pointer?

`ssd1306 = new Adafruit_SSD1306(128, 32, &Wire);`

vogt31337, 2020-09-06:
Hmm sounds reasonable, I'll try it and if it works I'll close the issue.

mikeysklar, 2026-10-30:
This compiles again since 2019, when WIDTH/HEIGHT lost const.
It now runs and corrupts memory: the temporary frees the canvas buffer.
PR #534 makes copying a canvas a compile error again.

Use a pointer, as BillyDonahue suggested:

```cpp
GFXcanvas16 *canvas = new GFXcanvas16(128, 32);
```

Display libraries like SSD1306 need the same change in their own repos.

I used Claude Code to test this.

closed 2026-10-30 by mikeysklar